WEC-Sim/WEC-Sim #543: Viscous Drag & Characteristic Area for WecSim's OSWEC

timmaflanagan, 2021-03-17:
Hi @kmruehl! I hope all is well with you! I am trying to define the drag coefficient for the tutorial version of the OSWEC that WEC-Sim has created. 

currently I have:

viscDrag.cd = ([1.17 0 0 0 1.17 0]); % this is a C_d value found online for a flat plate where the flow direction comes from x. 
viscDrag.caracteristicArea = ([160.2 0 0 0 934.2864 0]); 
the characteristic Area in x is calculated using 18*8.9,
the characteristic Area in Rx calculated using 18*8.9*(characteristic length)^3
where (characteristic length)^3 = (V_body/A_surface)^3=((18*1.8*8.9)/(18*8.9))^3

In post #130 you mention that there is literature to explain what the characteristic area and C_d values are. Could you verify that these numbers are correct and please let me know where to find the values for comparison?

Thanks,

yuyihsiang, 2021-03-17:
You can use two different approaches. One by specifying the viscous damping as you have defined but the other approach is to use Morison Element (see the attachment and the Desalination example in the WEC-Sim_Application Repository)
 
[Yu, Jenne - 2018.pdf](https://github.com/WEC-Sim/WEC-Sim/files/6160056/Yu.Jenne.-.2018.pdf)

timmaflanagan, 2021-03-17:
Hey YiHsiang,

Thanks for this info. It is helpful to have values for the Morison element on hand!

closed 2021-03-17 by timmaflanagan