WEC-Sim/WEC-Sim #544: Questions: Rectifying Valve, Hydraulic Motor, Wave Elevation VS Absorber Power PistonNCF

rodrijardim, 2021-03-18:
Hello everyone,

I am working on the examples already provided by the program, namely, "RM3_Hydraulic_PTO".
[RME_Hydraulic_PTO.zip](https://github.com/WEC-Sim/WEC-Sim/files/6167102/RME_Hydraulic_PTO.zip)

After the simulation and visualization of the results provided by the file "userDefinedFunctions.m" I had some doubts.

**1) Rectifying Valve** = As I understand it, this valve allows you to rectify the direction of the hydraulic flow (converts bidirectional fluid into unidirectional). The result of the "volFlow" is shown in the image below.
- **1a) What does "volFlow" mean and what is it for?**
[image]

**2) Hydraulic Motor** = Is a mechanical actuator that converts hydraulic pressure and flow into torque and angular displacement.

- **2a) What the outputs "delPress", "motorVol", "volFlowMHigh", "volFlowMLow" mean.**
- **2b) What is the purpose of the "delPress" and "motorVol" inputs on the rotational generator?**
[image]

**3) "output.ptos.forceActuation" VS "output.ptosim.pistonNCF.force"** = Why does it have the same result? I thought that the strength of action in the PTO had the contribution of all "components" present in it.
[image]

I apologize if any of my doubts have already been answered or if they are present in the documentation. Thank you in advance for your answers.

jleonqu, 2021-03-24:
Hi @rodrijardim

Before answering your questions, please consider that this example is not taking into account the compressibility of the fluid.

1) volFlow is the volumetric flow in the hydraulic cylinder. The flow input in the cylinder is equal to the flow output since the flow is considered non-compressible. In the rectifying valve it is necessary to apply the conservation of mass, which in this case means that the flow in the high pressure line is equal to the flow in the low pressure line.

2) delPress is the delta pressure between the high pressure and the low pressure side of the motor
motorVol is the volumetric displacement of the hydraulic motor. There is a function called variableMotorVolume.m that calculates the displacement as a function of delta pressure.
volFlowHigh is the flow from the high pressure side of the motor
volFlowLow is the flow from the low pressure side of the motor

The purpose of the "delPress" and "motorVol" inputs is to calculate the hydraulic torque from this equation: Hydraulic Torque = delPress*motorVol. That goes to the generator because this term is part of the dynamic equation for the generator shaft.

We are currently working on some improvements for the PTO-Sim blocks in order to make them more useful, versatile, and easy to use.

3) "output.ptos.forceActuation" takes into account all the forces on the PTO, including damping forces and the input force that comes from the PTO blocks. In this case, the damping coefficient is set to zero, so the only force acting on the block is the force from the hydraulic PTO, that is why these two "output.ptos.forceActuation" and  "output.ptosim.pistonNCF.force" are equal.

rodrijardim, 2021-03-25:
Hello @jleonqu, thanks for the help!

I have 2 more questions:

     1) I assume that the "volFlow" unit is m ^ 3 / s and the "motorVol" unit is cm ^ 3? Is correct?

     2) I would like to know more about the function present in "variableMotorVolume.m". Where does the WEC-Sim team get these values for displacement? Is there any documentation on this?

Thank you !

jleonqu, 2021-03-26:
Hi @rodrijardim 

1) "volFLow" units are m^3/s and "motorVol" units are m^3/rev.

2) The file"variableMotorVolume.m" changes the displacement of the hydraulic motor as a function of the pressure using a linear model. In this case, the function is simulating a variable displacement hydraulic motor, but you can change that function depending on your specific needs. For example, if you are simulating a fixed displacement hydraulic motor, then you can change that function to make the output a constant, or you can just delete the Matlab function block and you can add instead a constant block with the value of the displacement for the hydraulic motor.

closed 2021-03-26 by rodrijardim