WEC-Sim/WEC-Sim #545: Latching Control

rodrijardim, 2021-03-23:
Hello everyone,

I am trying to implement locking control in the "RM3_cHydraulic_PTO" model to "lock and release" the float at the top and bottom point of the wave in order to impose greater pressure and speed on the PTO. I made some changes to the "PTO actuation force of translation", as you can see in the image below
[image]

After simulating the model, I verified that the PTO does exactly what was planned as you can see in the figure below (PTO position vs. wave elevation):
[image]
Then I visualized the Power Absorbed by the PTO VS Electrical Power VS Mechanical Power and the Force Actuation PTO. I compared these results with the results of the same model, but without the latching control and found that they are exactly the same.
[image]

I thought that the model with locking control should have a greater PTO Actuation Force and also a greater production of electric energy. Am I thinking wrong?
[LatchControl.zip](https://github.com/WEC-Sim/WEC-Sim/files/6191126/LatchControl.zip)

Thank you!

dforbush2, 2021-03-24:
Hi @rodrijardim 

I think that the issue has to do with how you put your control block in the model. With how you have included the latching control into the model, the actuation force imposed by this controller will NOT be included in the PTO actuation force field, and depending on how you are calculating power capture, will not have an affect there either. It seems you have defined latching control as more of an intermittent kinematic constraint than an actuation force.

I would recommend you restructure your simulation to use this block as your translational PTO: 
[image]

This way, the actual actuation force applied will be calculated by the PTO block, and included in your resulting power calculations.

jleonqu, 2021-03-24:
Hi @rodrijardim 

Another option could be to implement your control algorithm in your subsystem "Hydraulic Motor and Generator". Your algorithm would receive as an input the response signals (position and velocity) and it can create as an output some kind of restriction in the hydraulic system in order to have the desired behavior.

rodrijardim, 2021-03-25:
Hello @dforbush2!

> It seems you have defined latching control as more of an intermittent kinematic constraint than an actuation force.

The value of the actuation force is calculated using the PTO "positon" and "velocity" present at the "response" output. This output goes to the "Hydraulic Motor and Generator" block, more precisely in the "Compressible Fluid Piston" block. This block receive the PTO "position" and "velocity" and calculates the "Force" which will then enter the "Force" input present in "Translational PTO Actuation Force". I think that this way I am changing the actuation force according to the position and speed of the PTO or not?
[image]

I was thinking during this day and assume the problem of obtaining the same result that the model without LatchControl has to do with the accumulators. The accumulators will be decharging practically the same flow to the hydraulic motor regardless of the highest speed reached in the rise and fall of the PTO.

Anyway, I will try to use the block that recommended me!

Thank you for your help!

rodrijardim, 2021-03-25:
Hello @jleonqu,

I will try to do what you recomended !

Thank you for your help.

closed 2021-04-02 by rodrijardim