WEC-Sim/WEC-Sim #546: Matlab R2021a bug - mask initialization

akeeste, 2021-03-23:
There seems to be a change in when Matlab R2021a handles mask initialization in Simulink. There are currently displacement, velocity and acceleration goto tags in each body block. If a user creates multiple bodies, the mask initialization takes the number input by the user, and updates the names of all relevant numbered blocks using that number.
So vel1, disp1 become vel2, disp2, etc for body #2.

It seems Matlab is now (in 2021a) checking for duplicate goto tags before running the mask initialization. There maybe be a way around this, but it is difficult to know Simulink's order of operations behind the scenes. The easy solution is to report a bug to Mathworks and wait for it to be patched. R2021a was just released 3/15/21, so this version should not be widely used and hopefully patches are released quickly. I'm making this issue so that WEC-Sim users/dev team are aware that R2021a is not currently usable.

I will submit a Mathworks issue on this and if necessary look into ways around this bug.

Adam

akeeste, 2021-03-29:
This issue is actually not a bug on the Mathworks side, but on how we check for version compatibility in the hydro/nonhydro/drag/flex body block masks. In R2021a, the Matlab version becomes v9.10. However, when this is converted to a numerical value it becomes '9.1' which incorrectly passes our version test of being '<9.7' and a nonexistent parameter is set. This led to an error in the mask initialization, which stopped early before renaming the goto tags. This gave a misleading error message related to the goto blocks but not the mask initialization error.

The version check is changed to:
ver = split(version,'.');
ver = str2num(ver(1:3));
if ver(1) < 9 || (ver(1)==9 && ver(2)<7)
   set_param(h,'ExtGeomFileType',geomType);
end

from the previous check:
ver=version;
ver=str2double(ver(1:3));
if ver<9.7
   set_param(h,'ExtGeomFileType',geomType);
end

mberk06, 2021-04-01:
@akeeste what's the schedule for releasing the fix? No rush, just excited to keep exploring the library.

akeeste, 2021-04-01:
@mberk06 

This is a small fix, but we want to have all changes to the master branch reviewed by a different team member to prevent creating another issue. Hopefully this can be implemented next week.

akeeste, 2021-04-13:
@mberk06 this R2021a bug should now be resolved! Please reopen this issue if you have any problems.

closed 2021-04-13 by akeeste