WEC-Sim/WEC-Sim #547: Heave Decay Test

ChrisJohnsonJ, 2021-03-24:
Hi, 
I am new to WEC-Sim and have been trying to validate the decay test performed in [Ocean Energy Systems Wave Energy Modelling Task: Modelling, Verification and Validation of Wave Energy Converters](https://www.researchgate.net/publication/336810450_Ocean_Energy_Systems_Wave_Energy_Modelling_Task_Modelling_Verification_and_Validation_of_Wave_Energy_Converters) 

I utilized NEMOH as the BEM solver and the structure was designed using the specifications given in the paper (sphere of radius 5m with a CoG(0,0,-2). However, when I ran the decay test, I am getting a result as displayed in the image
[image]
 
I know this is not the right result as I expect the similar results as displayed in the paper. I would really appreciate it if you can guide me to reach the appropriate result as in the paper. 

Thank you!

akeeste, 2021-03-24:
Hi @ChrisJohnsonJ 

This is certainly strange behavior going on. Can you please upload your case directory? With the case set-up, we are able to look and see what may be going on with your WEC-Sim issue.

Also, this IEA OES Task 10 case is actually an application on the [WEC-Sim Applications repository](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Free_Decay). If you have not already, I highly recommend viewing this WEC-Sim MCR case as it can guide you to replicating the results. The sphere is tested in free decay for initial displacements of 0m, 1m, 3m, and 5m. 

Adam

ChrisJohnsonJ, 2021-03-25:
Hi @akeeste,

Thank you for your reply. I have attached the case directory with this message. I hope that it helps give you get a better picture of the issue I have. 
[Test 14.zip](https://github.com/WEC-Sim/WEC-Sim/files/6203352/Test.14.zip)

I had already used the application for the IEA OES Task 10 from the [WEC-Sim Applications Repository](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Free_Decay). Unfortunately, I get the same weird result. I am suspecting my issue orginates from calculating the Hydrostatics utilizing NEMOH.  Please do let me know if you need more information. 
I look forward to your reply.

Thank you so much!
Chris

akeeste, 2021-03-26:
Hi @ChrisJohnsonJ,

I have been taking a look at this case, and I would agree that the issue seems to be with the hydrostatics and other BEM data. It seems that in your case, the displaced volume and hydrostatic stiffness are roughly double the application values. This is giving the equilibrium position of ~1.75m in the above plot, instead of zero. This is also likely way the above plot appears to have a phase change. There likely isn't an actual phase difference in the 0m, 1m, 5m cases; it's just that the 5m case decreases to 1.75m at equilibrium while the 0m,1m case increase to equilibrium at 1.75m.

I am not an expert in Nemoh, but I would first check the mesh position and resolution. If the cg is offset in the hydrostatics, it may be accounting for the entire hemisphere instead of the smaller submerged portion. Also, I am unsure if the figures /Refined/Fig 1.fig, Fig 2.fig, etc represent your current BEM solution, but they do differ from the application BEM data. If you have not already, I suggest refining your mesh further and checking for convergence. Though this could be from differences between WAMIT and NEMOH as well.

Let me know if you have more questions on this! I have not run the case with adjusted hydrostatics, but suspect this will solve your issue.

Adam

ChrisJohnsonJ, 2021-03-27:
Hi @akeeste ,

Thank you for your reply!
I learned that the issue occurred with me incorporating the entire hemisphere instead of half of the hemisphere while calculating the hydro statics using NEMOH. I was able to get the correct solutions for the BEM data and was able to run the decay test and generate the right solution as attached below.

[image]

Thank you for your guidance and your prompt replies!

Regards,
Chris

akeeste, 2021-04-01:
Yes the application repo does show oscillations dying out between 25-40s
[image]

How does your BEM data now compare to that in the applications repo?
I would first check the standard plots from BEMIO (added mass, radiation damping, etc in heave). Especially, look at how the excitation IRF compares between your BEM data and the application as this is the forcing used for the noWaveCIC and regularCIC cases.

Adam

yuyihsiang, 2021-04-07:
[Sphere-NEMOH.zip](https://github.com/WEC-Sim/WEC-Sim/files/6274842/Sphere-NEMOH.zip)

The attachment is the hydrodynamics coefficients obtained from Nemoh from the IEA OES Task 10. You can use that to compare to your Nemoh as well.

--Yi-Hsiang

ChrisJohnsonJ, 2021-04-08:
@yuyihsiang 

Thank you so much for sharing the files Yi-Hsiang!

I was able to make certain changes to my files using the files your shared. 
However, I ran into a new issue. If I run NEMOH with meshes more than ~3000 panels, NEMOH abruptly stops while Solver.exe is running. I tried running it on different computers but I am encountering the same issue. 
The NEMOH procedure I have used till date is as follows;  SALOME(for Meshing) -> Exporting and Converting using NEMOH Converter -> Inputting parameters in Nemoh.cal file -> Copying the necessary .exe files for NEMOH in the case directory -> run Nemoh using a .bat file which starts the preprocessor, solver and postprocessor -> Using Mesh.m to generate Hydrostatics.dat & KH.dat

I searched through the issue history in the forum, but I have exhausted all methods. It would be kind of you if you could help me resolve this. 

Thank you!

-Chris J

ChrisJohnsonJ, 2021-04-09:
@yuyihsiang @akeeste 

Dear Yi-Hsiang & Adam,

I have some good news. I created the meshed structure using the matlab routines and running NEMOH using the Matlab wrapper and have solved the issue. I ran the Decay test using the HDF5 file created using the above routine and got the appropriate results as attached below.
[image]

The main mistake I made was using the SALOME exported mesh file which contained more than 2000 panels, where NEMOH abruptly stops. I read that max panels NEMOH uses initially is 2000, after which I could refine the mesh while running axiMesh.m. 
Using the aximesh.m Matlab routine directly _to create the mesh_ instead of using SALOME proved to be more easier and efficient than the procedure I was using earlier since I didn't have to go through the hassle of meshing in SALOME and converting it into NEMOH readable format using the converter bash file. 

Thank you for your guidance and your support!

Chris

closed 2021-04-09 by ChrisJohnsonJ