WEC-Sim/WEC-Sim #549: Multiple Gotos when running the RM3 Example

mberk06, 2021-03-29:
I just started working with the WEC-Sim library and am new to Matlab. Really cool so far!

I'm on the last step of the tutorial [run wecSim in the RM3 dir](https://wec-sim.github.io/WEC-Sim/man/getting_started.html#step-3-test-the-installation), but get the following error:

```
List of Constraint(s): Number of Constraints = 1 

	***** Constraint Name: Constraint1 *****

Simulating the WEC device defined in the SimMechanics model /Users/NAME/Documents/Marine_Energy/WEC-Sim/WEC-Sim/examples/RM3/RM3.slx...   
Error using wecSim (line 376)
Error due to multiple causes.
Caused by:
    Error using wecSim (line 376)
    Multiple Gotos found with tag named 'disp1'
    Error using wecSim (line 376)
    First offending Goto 'RM3/Float/Hydrodynamic Body/Goto'
    Error using wecSim (line 376)
    Second offending Goto 'RM3/Spar//Plate/Hydrodynamic Body/Goto'
```

Any idea why this could be occurring? Does it have something to do with [the newest version of matlab](https://github.com/WEC-Sim/WEC-Sim/issues/546)? I'm using `9.10.0.1602886 (R2021a)` on a mac. 

Thanks in advance!

P.S. all the issues described [here](https://github.com/WEC-Sim/WEC-Sim/issues/250) occurred for me because I didn't use lfs when cloning. I fixed it, but just in case you guys get similar issues again you can add that to the debug steps.

akeeste, 2021-03-29:
Hi @mberk06 

Yes we are not currently compatible with Matlab R2021a. I'm working on a fix with Mathworks, details of which are in issue #546 . My current thought is that Simulink's order of operations changed from R2020b to R2021a. It seems that now the error checking for multiple Goto tags is performed before initializing the masks that update them with the correct number for the body, constraint, pto, or mooring.

Thank for pointing out the git lfs issues. I'll look at adding those steps into the 'getting started' documentation sections more clearly.

If you have any more feedback on the R2021a issue while we debug it, feel free to comment on issue #546 . We can keep the discussion consolidated in that issue for future reference.

Adam

mberk06, 2021-03-29:
@akeeste great will do. And thanks for the help!

closed 2021-03-29 by akeeste

antbog, 2023-06-01:
Hi! I'm facing the exact same problem with MatLab R2023a right now. @mberk06 what did you do exactly to make it work ? 
Thanks in advance