WEC-Sim/WEC-Sim #550: Motor characteristics on PTO-Sim

SofiasKonstantinos, 2021-03-31:
Hello everyone , 
As i continue my project i would like some further guidance on the PTO-Sim . I cant seem to find much documentation on the characteristics of the generator and how it is described in matlab . I found in the wec-sim site this phrase "The datasheet in tutorials is taken from ABB datasheet part number M3BJ315SMC" which then i looked up but didnt quite get results . Any ideas on documentation or some guidance on what i need to know about this topic .

akeeste, 2021-04-01:
Hi @SofiasKonstantinos 

PTO-Sim is fully described in So, et al. 2015, which is reference 7 in the [documentation](https://wec-sim.github.io/WEC-Sim/man/publications.html#wec-sim-publications). Likely, an ABB motor, series M3BJ was used as a sample for these applications. However, I would stress that the PTO-Sim applications are only examples. They are meant to guide users to implementing their own PTO designs, not serve as standards. As stated in the PTO-Sim/RM3 Hydraulic PTO [documentation](https://wec-sim.github.io/WEC-Sim/man/advanced_features.html?highlight=pto#tutorial-rm3-with-pto-sim):

``
Step 8: If a rotary generator lookup table is used, this block assumes the user will provide the datasheet. After the datasheet is 
loaded into Workspace, it needs to be named as table because the word table is used inside Simulink lookup table block. The 
datasheet in tutorials is taken from ABB datasheet part number M3BJ315SMC
``

I would recommend substituting the given motor datasheet for an efficiency look up table you are familiar with. The motor table should some arbitrary torque and angular frequency value to a power efficiency.

[image]

@jleonqu has been working to revamp PTO-Sim and may be able to give you more insight on its future use.

Adam

jleonqu, 2021-04-01:
Hello @SofiasKonstantinos 

I want to give you a brief description of the generator block.

Inputs:
delPress: Pressure difference between the high pressure line and the low pressure line
motorVol: volumetric displacement of the hydraulic motor. There is a function called variableMotorVolume.m that calculates the displacement as a function of delta pressure.
angVel: angular velocity of the shaft

Output:
genTorque: torque in the generator. This torque is an input for the hydraulic motor block, where the dynamic equation for the shaft is solved.

The torque in the generator is calculated with the next equation: genTorque = (delPress*motorVol/(desiredSpeed*1.05))*angVel
Where that 1.05 accounts for the mechanical efficiency and the term angVel/desiredSpeed accounts for the volumetric efficiency.

In the WEC-Sim we are currently working on some improvements for these blocks to make them more easy to use.

jleonqu, 2021-04-09:
Hi @SofiasKonstantinos 

Does this information answer your question?

SofiasKonstantinos, 2021-04-09:
Yes this information was helpful , I still have some work to do but as i continue i will reach out to you again for further guidance . Thank you very much once again .

jleonqu, 2021-04-19:
Hello @SofiasKonstantinos 

The variable omegapu is defined as the ratio between the actual angular velocity and the rated angular velocity defined in the ptoSimInputFile as ptosim.rotaryGenerator.omegaBase. Then, the angular velocity in the motorEff.mat file goes from 0.004*ptosim.rotaryGenerator.omegaBase to 4*ptosim.rotaryGenerator.omegaBase. The units for ptosim.rotaryGenerator.omegaBase are rad/s. Something similar happens with the definition of the torque, in this case the torque changes from -1.56*ptosim.rotaryGenerator.TgenBase to 1.56*ptosim.rotaryGenerator.TgenBase. The units of ptosim.rotaryGenerator.TgenBase are N*m

SofiasKonstantinos, 2021-04-19:
So to just to make it clear in my mind , for the RM3 example the ptosim.rotaryGenerator.omegaBase is 300 , so my omega range is practically 1.2rad/s to 1200rad/s . And accordingly the torque is -3125Nm to 3125Nm . This means that the maximum power outpout of this generator is 3125*1200=3750000Watts or 3750Kw . When i run the simulation for a specific wave Height and peak period i get a variable outpout.ptosim.rotaryGenerator.elecPower , i imagine this is in Kw ? I am trying to create bins for a power matrix but cant seem to get it right . Also the ptosim.rotaryGenerator.desiredSpeed is set to 150rad/s is it a random number for the example or is it realistic ?

jleonqu, 2021-04-23:
Hello @SofiasKonstantinos 

The units for output.ptosim.rotaryGenerator.elecPower are Watts, and the value of the desired speed is just an example.
I took the information of the efficiency table that is used for that specific example and I did the efficiency contour plot and the power contour plot:

Efficiency plot:
[image]

Power contour plot:
[image]

The maximum power from the power contour plot is around 800 kW.

jleonqu, 2021-05-05:
Hello @SofiasKonstantinos 

Does this answer your question?

SofiasKonstantinos, 2021-05-25:
[image]
[image]
[image]
[image]

Dear @jleonqu sorry for not responding in time ,
This is a power matrix a created with the inputs from the images attached above . I dont seem to get any close to the 800kW range .. but i can't find why

jleonqu, 2021-06-02:
Hi @SofiasKonstantinos 

800kw is the maximum power reported for the electric machine used in the example, but that doesn't mean that the case you are running is going to reach that value. If you want to increase the power reported in the table, I would suggest to change the characteristics of your hydraulic PTO such as the area of the piston, the displacement of the hydraulic pumps, the sizes of the accumulators, etc.

jleonqu, 2021-06-16:
Hi @SofiasKonstantinos 

Does this answer your question? Let us know if we can help you with anything else.

closed 2021-06-23 by yuyihsiang

hossein-mehdipour, 2021-09-26:
Hello @jleonqu ,
I wanted to know the rated power of the generator used in the HPTO case for RM3 (for non-compressible fluid). Am I suppose to calculate it just like you did with different inputs to find out the maximum power I can reach. Or is it a factory property that I should search and find.
Thank you in advance

jleonqu, 2021-10-20:
Hello @hossein-mehdipour 

Let me just write again the equation because I think the multiplication symbols were not included by the text editor: genTorque = (delPress*motorVol/(desiredSpeed*1.05))*angVel

This equation comes from the torque equation for a hydraulic motor. The PTO-Sim blocks were created having in mind one specific example, where a hydraulic motor is connected to an electric motor. So, for this block, the generator torque is equivalent to the hydraulic motor torque. References 7 and 10 [here](http://wec-sim.github.io/WEC-Sim/master/introduction/publications.html) are related to PTO-Sim development.