WEC-Sim/WEC-Sim #554: Singularity Problem - 2-Body Attenuator WEC

7jcapper7, 2021-04-09:
I am trying to simulate a 2-body attenuator WEC to determine the power absorption. I want each body to move in heave and pitch only, so I am using a floating 2DOF constraint. I have a rotational PTO between the two bodies. I can get some results for regular waves, but I get a singularity error message when using irregular waves. I have tried a smaller time step and tried varying the wave frequency, but still cannot get the model to work. Can someone please help me?

[image]
[image]
[image]

[2 Body_CJ_FP_PC8_3DOF.zip](https://github.com/WEC-Sim/WEC-Sim/files/6287310/2.Body_CJ_FP_PC8_3DOF.zip)

akeeste, 2021-04-12:
Hi @7jcapper7 

Thanks for uploading the case files immediately. I can take a lot at your issue and help you out. While I look at your case, I would say that this singularity issue usually comes up when there is not a stable mass/displaced volume for the system, or are issues in the BEM input data.

As a starting point for any new case, I would recommend running the following test waves in order:
noWave
noWaveCIC
noWave + initial displacement
noWaveCIC + initial displacement
regular
regularCIC
...
irregular wave

This will help you in debugging where the issue is occurring. For example, if there are stability issues without CIC but not when using CIC, then there are likely irregular frequency issues in the BEM data. If there are issues with the noWave or noWave+initial displacement cases, the hydrostatic solution is not stable. If a noWave case works but not a regular case, there is a problem with BEM data or drag values.

Adam

akeeste, 2021-04-12:
@7jcapper7 

A couple notes - 
Even though the case is unstable, I do think your current Simulink set-up is more reliable than the others in the powerpoint file. It will be simplest and easiest to have one flap heave/pitch at the joint, and allow the other to pitch about the same joint. This effectively gives the heave and pitch for both flaps that you see in regular case that works.

I tested this case with the noWaveCIC case and the solution is still unstable. Before proceeding to any regular or irregular case, I highly recommend ensuring noWave, noWaveCIC and noWave+initial displacement cases are stable before proceeding. First I recommend rerunning your BEM solution. BEM data at higher frequencies is needed to ensure that the IRF decays without oscillation, and that the current infinite frequency added mass is calculated. Ensure that the first peak frequency is captured.

As noted in the BEMIO plots, added mass should decay to a constant, radiation damping should decay to 0 and the excitation force should level of as well:
[image]
[image]
[image]

7jcapper7, 2021-04-12:
Ok thanks I will give it a try.

7jcapper7, 2021-04-12:
I reran the BEM solver (Ansys AQWA) to include data up to 1 s wave period (6.283 rad/s frequency). You can see the new plots below. It does not look like the plots are displaying the desired behavior (decaying to constant A and zero B). Do you have any recommendations?

[image]

[image]

[image]

akeeste, 2021-04-12:
@7jcapper7 In your previous BEM data (and it appears here too) the AQWA frequencies are heavily weighted towards 0 rad/s. I would space your frequencies linearly, or add additional points from 2-6 rad/s to get smooth data. Also, if you can add a lid or run your device with an irregular frequency removal option, do that as well to prevent spikes in the results.

Adam

7jcapper7, 2021-04-15:
Do you know any good resources explaining how to create lids in AQWA? I have been looking at the AQWA user manual, but it does not go into a lot of detail.

kmruehl, 2021-04-15:
@7jcapper7  it's been a very long time since I've used AQWA, but the workflow was thin, freeze, slice, then mesh. The thin makes a surface body, and the slice cuts the body at the SWL. That way when you mesh the device you have an internal lid for the free surface. I imagine the workflow is the same now.

7jcapper7, 2021-04-15:
I am actually trying to create an external lid in the gap between the two bodies. I am wondering if maybe there is hydrodynamic interaction occurring in the gap that is leading to unrealistic AQWA results.

kmruehl, 2021-04-15:
@7jcapper7 I see, well then that's entirely different. Once again, it's been years since I've used AQWA, but I recall there was a setting in WorkBench to add an external lid. I'm not familiar with the AQWA documentation anymore, but I recommend contacting ANSYS for this request.

akeeste, 2021-04-21:
Hi @7jcapper7 

A couple more comments: 
- When evaluating the quality of the BEM data, I also recommend looking at the excitation force IRF. If you BEM data does not span enough frequencies with enough resolution, the IRF will be very erratic and not decay to zero in a reasonable amount of time.
- I also recommend setting the cg to below the still water line initially. If the cg is at the SWL you might run into some roll or other rotational instabilities that can cause the singularity issue as well with larger waves.

Does this sufficiently answer your question about about the singularity issue? Unfortunately we cannot give more support on using AQWA. 

Adam

7jcapper7, 2021-04-21:
What do you expect as a reasonable amount of time for the IRF to decay? I have been trying to implement your suggestions, and you can see my most recent IRFs below.

[image]
[image]

akeeste, 2021-04-22:
Hi @7jcapper7 

Your IRF plots do look reasonable now, it looks like a CI Time of 60-100s would work fine. Does this new BEM data result in a physical response for the device in noWaveCIC and regularCIC waves? 

These two wave types use the same convolution integral for radiation damping that irregular waves do. If these cases are stable, you can likely run your device in an irregular case without the original issue.

7jcapper7, 2021-04-22:
How did you know to use a CI time of 60-100 s? It looks like 60 s is the default.

For noWaveCIC, the motion quickly decays to zero. For regularCIC, the model can run fine without crashing. I can get some results for irregular waves now. The added mass and radiation damping BEM plots are looking better, although I am unsure about the excitation force magnitude.

noWaveCIC: 
[image]

irregular wave:
[image]

[image]

[image]

[image]

akeeste, 2021-04-23:
@7jcapper7 

That is good that the nowaveCIC and regularCIC cases are now working!

With regards to the CI time, I estimated that by the time it takes for the IRF to decay to zero without oscillation. Based on the plots you uploaded, this seems to be any time greater than 60s.

The last action I recommend taking for your BEM data is to conduct a mesh resolution study. Run your device in AQWA with an increasingly fine mesh until the results converge. If you need to rerun in the future with different frequencies or directions, use the fastest simulation that gives converged results.

Adam

akeeste, 2021-04-28:
Hi @7jcapper7 

Has this solved your singularity issue? Unfortunately we cannot support BEM simulation and output anymore. If there are still issues once fixing the BEM please give updates and we can discuss from there.

7jcapper7, 2021-04-29:
The model seems to be running pretty well now. Thank you for the help.

closed 2021-04-29 by akeeste