WEC-Sim/WEC-Sim #555: A question about the calculation of wave amplitude in wave spectum

linmeihong, 2021-04-11:
In the function irregWaveSpectrum(obj,g,rho) of 'waveClass.m', I found:
(1)
obj.S = S_f./(2*pi); 
obj.A = 2 * obj.S;
(2)
in the WECSim_Lib/../../Irregular Wave Excitation Force
F_wave = sum(A*cos(w*t) * Re{Fext(i)} - A * sin(w*t) * Im(Fext(i)} + A * Re{Fext.md(i)}), where i = each frequency bin
(3)from the WECSIM document: Section Convolution integral formulation
F_exc(t) = Re[R_f(t) sum{F_exc(wj,xita) e^{i*wj*t+phasej}  sqrt{2S(wj)dwj})]

from (1) A = S_f/pi 
but from (3) A = sqrt{2S(wj)dwj}, this equation can also be found in AQWA Theory manual version 19.0 (formula 2.20)

May be the obj.A is computed in other place? can any one tell me where the malab code for computing the wave amplitude?
thank you very much for your help.

nathanmtom, 2021-04-12:
@linmeihong Thank you for your question. The obj.A is computed in WEC-Sim as you have listed above. If you look in the MATLAB FCN block in the WEC-Sim library you have listed above, you will see on line 13 that a square root is taken to be consistent with Eqn. (3) you have listed above. 

Cheers, 
Nathan
National Renewable Energy Laboratory

linmeihong, 2021-04-12:
thanks a lot , nathanmtom. I find it. 
 C1 = sqrt(bsxfun(@times,A*WaveSpread(ii),dw));
from the source code, I learn so much for the implementation of theory. thanks.

nathanmtom, 2021-04-12:
@linmeihong Glad that our suggestion helped find the information you needed.

closed 2021-04-12 by nathanmtom