WEC-Sim/WEC-Sim #558: OSWEC AEP or Power Matrix

AleixWG, 2021-04-14:
Hi

Im trying to estimate the AEP for a "standard" flap, the one you have in the tutorial / example repository, so i tried to run a number of simulations using your .slx and .h5 in a fresh WECSIM, downloaded and unziped today, the model works perfectly fine (unless I modify the damping control, when it just go crazy), but I'm somehow surprised about the results, they are really low.

Using the default parameters for PM irregular waves (Hs 2.5 - Tp 8), if I check the output power internal mechanics (:,5) and calculate the average I obtain 45 Watts. So, I'm wondering if I'm missing something for this particular technology when it comes to power generation estimation or I'm completely failing to understand how to calculate the AEP (or Power Matrix cell) here. Attached the results in .mat format so you can maybe compare with another simulation of your own.

[Results_WG.zip](https://github.com/WEC-Sim/WEC-Sim/files/6312545/Results_WG.zip)

Thanks in advance

nathanmtom, 2021-04-16:
@AleixWG Thank you for your question and will do our best to answer. 

First, to your question about damping control I'll need a bit more information on what you did to adjust the PTO damping and more details on what the irregular or unexpected motion profile looks like. 

To your question about low power performance: It appears you are using the input file from the examples folder where we have implemented directional spreading in the waves (see lines 27 and 28 in the wecSimInputFile.m from the examples -> OSWEC). The percentage of the waves coming parallel to the front device is greater than coming perpendicular where you would generate the most rotational motion. Therefore, I suggest first deleting lines 27 and 28 and re-run with the default PTO damping coefficient and recalculate the average power output. That should increase the average power, but you can likely optimize the power further by iterating across a wider range of PTO damping coefficients to find the optimum. 

Please let me know if this suggestions improve your results to your satisfaction. If not, please provide additional comments or description of the problem you are encountering. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

AleixWG, 2021-04-20:
Thank you for the time and effort in your reply.
I deleted line 27 and 28, we improved a bit the average power from 45 Watts to 243 Watts.
But the point was related with the control, i was expecting a good average power output with the default settings in the example, my bad, now i get around 200 kW average with only 3 iterations of damping.

I would like to ask something else, this is a more generic question about the Capture Width Ratio:
In WECSIM you calculate the power of the wave per meter crest in the variable Pw, for deep water and irregular waves:

obj.Pw = sum(1/2 * rho * g^(2) * S_f .* obj.dw ./ obj.w); 

Now i have to name some references:
https://hal.archives-ouvertes.fr/hal-01145072/document -> Here, they use results from multiple sources, one of them is the RM3, (quick view in table 8, page 32). They use the following information:
- CWR = 16%
- Pw = 34 kW/m
- D = 20 m

Cited literally: "[27] is the final report of a technico-economical study for four marine renewable energy technologies: three current turbines and one WEC (RM3). It is a heaving WEC, inspired by the OPT technology. The power matrix was determined using numerical modeling and electricity production calculated for a site offshore California. The wave resource is 34 kW/m. The numerical model was validated against experiments. Electricity production is 700 MWh at this site, assuming 80 % PTO efficiency, 95 % availability and 98 % efficiency in the transmission line. Thus, the mean absorbed power is 108 kW. The diameter of the buoy is 20 m, thus the mean annual CWR is 16 %."

Ok, i want to focus in the wave resource information, they use 34 kW/m, i used the same parameters (scatter diagram) as in https://energy.sandia.gov/wp-content/gallery/uploads/SAND2014-9040-RMP-REPORT.pdf, and the only way I obtain 34 kW/m is using the "average" power distribution for irregular waves Pw = 0.5 * Hs^2 * Tp. If I use the Pw output from wecsim and calculate the wave energy matrix average it results 177.22 kW/m

Any idea why this variation?

nathanmtom, 2021-04-20:
@AleixWG Apologies for misunderstanding you initial question, but it sounds like you have still found your answer. 

To you latest question, what hydrodynamic file are you using to complete the scatter diagram you are referencing? The power contained within a sea state is dependent not only on the peak period and significant wave height, but also the water depth. In Sec 5.2.1.2 of the second referenced report, the shown equation makes the approximation that is shown requires the water depth  to be considered deep. The RM3 files placed on the Github repository should be placed in deep water while the OSWEC model you have cited above will be in shallow water which changes the power contained within a propagating wave. This would only affect the output from WEC-Sim not your other calculations. 

Also, note that in WEC-Sim the spectrums are defined by the peak period of the wave and not the energy period of the wave as shown in Table 5-1 in the RMP report. Therefore, you need to be careful what inputs are going into your hand calculation against what is going into WEC-Sim for consistency. 

If you can review your calculations and WEC-Sim inputs to confirm or update my statements above we can further troubleshoot your question. 

Cheers, 
Nathan

AleixWG, 2021-04-21:
@nathanmtom 

Here we are using RM3 model, not OSWEC, maybe my comments are more clear with the attached document. Here we used WECSIM to try and validate the power matrix in the reference, the result is reasonable, variations probably because of the damping control, but we found that the available wave energy is completely off, and we don't have any control variable there, also 177.22 kW/m of avaiable wave power seems too high for any place.

Best,
Aleix

[RM3validation.xlsx](https://github.com/WEC-Sim/WEC-Sim/files/6349779/RM3validation.xlsx)

nathanmtom, 2021-04-21:
@AleixWG Thank you for that clarification and supporting documentation. After reviewing the calculation you have cited above, I've come to the conclusion that WEC-Sim is likely missing a 2*pi factor in its calculation. After accounting for the 2*pi factor and converting from peak period (Tp) to energy period (Te) there is better matching between your hand calculation and WEC-Sim outputs, see the updated spreadsheet. Please note that your equation listed above uses Tp in the energy calculation when it should be the energy period. The peak period is used as input to WEC-Sim, but not the energy period. 

There are still some differences between the WEC-Sim output and hand calculations which are most likely because WEC-Sim uses the frequencies imported from the .h5 file to calculate the wave spectrum so sufficient frequencies at the low and high frequencies are needed to account for the full frequency integration that provides the closed form solution you have cited above. In addition, the 0.5 factor you have included in your calculation rounds up from 0.49 (for sea water 1025 kg/m3) and from 0.478 (for water 1000 kg/m3) which can result in a few percentage points off in the final numbers. 

The WEC-Sim team will confirm and then correct for this issue and you can reference Pull Request #562 to track this update.

[RM3validation R1.xlsx](https://github.com/WEC-Sim/WEC-Sim/files/6354432/RM3validation.R1.xlsx)

AleixWG, 2021-04-22:
@nathanmtom This is great! Thank you Nathan, I completely agree with your comments, i should have modified the Tp to Te before sending the excel file, glad to find this missing 2pi factor.
I don't close with comment, just waiting for that confirmation from WEC-Sim team.

Best
Aleix

nathanmtom, 2021-05-03:
Resolved per Pull Request #570

closed 2021-05-03 by nathanmtom