WEC-Sim/WEC-Sim #559: Modelling a Vertical Axis pendulum in Simulink

ChrisJohnsonJ, 2021-04-20:
Hi, 

I am trying to model a vertical axis pendulum in a sphere using the Library blocks provided by WEC-Sim. I was wondering if there are any examples which I could use the model the pendulum(PTO) in Simulink. I would really appreciate any material regarding this or references which could help me have a develop the model. 
Thank you so much!

kmruehl, 2021-04-21:
@ChrisJohnsonJ Thanks for the question. Can you clarify what you mean by a vertical axis pendulum in a sphere? Can you provide an image/example? Thanks!

ChrisJohnsonJ, 2021-04-22:
Hi @kmruehl ,

Thank you for your reply! I have attached the image 
[image]

I am trying to model the WEC studied in this paper by Boren Blake
[https://ieeexplore.ieee.org/abstract/document/6859459]

I hope this gives you a better idea! Please do let me know if you need more information and I look forward to hearing from you soon!

Thank you!

kmruehl, 2021-04-30:
@ChrisJohnsonJ Sorry for the delayed response, I didn't see that you had responded until now. 

Sure, I can provide some guidance on how I would recommend modeling this device in WEC-Sim. I've provided a screenshot below for reference, and also attached an example Simulink model of this configuration. Here's what I recommend:

- Global Reference Frame for the SeaFloor
- `constraint(1)` 6DOF Constraint (or really any motion constraint) for the Mooring. Option to include MooringMatrix
- `body(1)` Hydro Body for the Float
- `body(2)` Non-Hydro Body for the Pendulum. Tthis is internal so it has no hydrodynamics, and thus no BEM data
- `pto(1)` Rotational PTO for the Generator/Gearbox. This block includes position actuation for control, but depending on the controller, you could use force actuation. For this block you will have to rotate its orientation relative to the global reference frame, because unfortunately the default for this block is rotation about the y-axis, for pitch motion, and you will need rotation about the z-axis for internal yaw motion. 

I hope this helps! Let me know if you have any more questions. 

[image]

[pendulum_wec.zip](https://github.com/WEC-Sim/WEC-Sim/files/6402509/pendulum_wec.zip)

kmruehl, 2021-05-05:
@ChrisJohnsonJ  does this answer your question?

ChrisJohnsonJ, 2021-05-06:
Hi @kmruehl, 

Thank you so much for your response and for the model file! I really appreciate it! I have a question regarding the dynamics behind it though. Since the PTO is fixed to the rigid body, wouldn't it mean there are two reference frames which would be required(one the gravity or global and one for the PTO) as the PTO moves along with the wave? I saw that the WEC-Sim blocks all work with one reference frame. How would this model take care of that? 

Would it be possible to get in touch with you via email? Since I wont be able to share some of my research material on the public platform yet. My email id is chrisjohnsontud@gmail.com.  
I look forward to your reply! Thank you so much!

Best Regards,
Chris

kmruehl, 2021-05-20:
@ChrisJohnsonJ WEC-Sim only requires one Global Reference Frame. This defines all bodies/joints/ptos/etc relative to a fixed global reference frame. The origin of WEC-Sim's Global Reference Frame, <0,0,0>, is at the SWL. Everything else in WEC-Sim is defined by a local, body-fixed, reference frame. In this case the PTO (`pto(1)`) is a rotational joint, defining relative motion between the Non-Hydro Pendulum Body (`body(2)`), and the Hydro Float Body (`body(1)`). Here's some [information from MathWorks about reference frames](https://www.mathworks.com/help/physmod/sm/ug/frames-and-frame-transforms.html), I hope this helps!

Unfortunately we cannot provide direct support via email. We ask all users to frame their questions in a way that they can be discussed on the WEC-Sim public forum for the benefit of the WEC-Sim user community. We encourage you to submit questions to the WEC-Sim forum, and we will do our best to provide support. Thank you

closed 2021-06-04 by kmruehl

yjkl12, 2025-12-02:
@kmruehl Hello, could you please take a look at the latest question I posted on my homepage