WEC-Sim/WEC-Sim #561: WEC RO simulation problems

veethan, 2021-04-20:
Hello all,

I am working on a project with wecsim, I ran some simple test with my simulink model first to verify the models and method, I am now trying to implement a RO desalination simulink system as the pto that was used in a previous project. However my issue is that when I try to run wecsim on my system, I get an error that refers to my accumulator. It states " Volume of liquid must be less than total accumulator volume." I have verified that this statement is satisfied in my RO system but it still throws the error after initialization. Any guidance is appreciated.
[WECsim_Buoy_project.zip](https://github.com/WEC-Sim/WEC-Sim/files/6346604/WECsim_Buoy_project.zip)

yuyihsiang, 2021-04-27:
I checked your model. Even for H=0m, the model still blows up. It might be helpful if you can provide some information how you determine the RO parameters.

yuyihsiang, 2021-05-05:
Do you still have the issue?

closed 2021-05-19 by kmruehl