WEC-Sim/WEC-Sim #563: how to define initial pitch angle for free decay test?

siya-jin, 2021-04-23:
Dear there,

I am trying to simulate a free decay test, setting the body with initial a pitch angle.
But the command of body(1).initDisp.initLinDisp = [ X Y Z], only offers the set for X Y and Z initial displacement. 
May i ask if i miss any command to control the initial angle setting for Roll, Pitch and Yaw, please?

Thanks!

siya-jin, 2021-04-23:
Dear there,

Sorry for the disturb. I thought i found the command of:
initDisp          = struct(...                                          % Structure defining the initial displacement
            'initLinDisp',          [0 0 0], ...             % Initial displacement of center fo gravity - used for decay tests (format: [displacment in m], default = [0 0 0])
            'initAngularDispAxis',  [0 1 0], ...             % Initial displacement of cog - axis of rotation - used for decay tests (format: [x y z], default = [1 0 0])
            'initAngularDispAngle', 0)  

to be used. 
Thanks!

akeeste, 2021-04-23:
Hi @siya-jin,

Yes you can use the initDisp structure to define an initial displacement of the device. You can set these values directly, or use the setInitDisp() method. This [method ](https://wec-sim.github.io/WEC-Sim/man/api.html#objects.ptoClass.setInitDisp) rotates a body about a specific axis and point, resulting in both rotation of the body and the corresponding translational displacement due to the rotation of the center of gravity about some arbitrary point.

siya-jin, 2021-04-23:
Hi Adam,

Really thanks for your help.

Best wishes
Siya

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

This email and any files with it are confidential and intended solely for the use of the recipient to whom it is addressed. If you are not the intended recipient then copying, distribution or other use of the information contained is strictly prohibited and you should not rely on it. If you have received this email in error please let the sender know immediately and delete it from your system(s). Internet emails are not necessarily secure. While we take every care, University of Plymouth accepts no responsibility for viruses and it is your responsibility to scan emails and their attachments. University of Plymouth does not accept responsibility for any changes made after it was sent. Nothing in this email or its attachments constitutes an order for goods or services unless accompanied by an official order form.

closed 2021-04-23 by akeeste