WEC-Sim/WEC-Sim #564: spectrumImport in WecSimMCR with different damping values

Pisdan96, 2021-04-24:
Hi there, 
I want to know if there is any way that we could differ the damping value with frequency if we import the spectrum data?
I changed the damping value when I used regularCIC with the help of .mat file. However, I could not find any way even in the repository.

I gave the spectrum data file as the mcrcasefile as given in the example

Best regards

dforbush2, 2021-04-28:
Hi @Pisdan96 

To answer your question I think I will need some more detail. It sounds to me like, at present, you are running an MCR case file that varies pto.c and waves.T, for a regular wave. You would like to do something similar, but where you are changing the waves.spectrumDataFile along with pto.c?

If this is the case, then you can make an mcr case file like so:

```
mcr = struct(); % initialize structure
mcr.header=[]; mcr.cases=[]; % initialize fields
mcr.header{1,1}='waves.spectrumDataFile'; 
mcr.header{1,2}='pto.c';
% list your spectrum files you want to import
mcr.cases{1,1} = 'YourSpectrumFile1.mat';
mcr.cases{1,2} = 'YourSpectrumFile2.mat';
mcr.cases{1,3} = 'YourSpectrumFile3.mat';
 % list the PTO damping values you wish to use
mcr.cases{2,1} = 100; % this is just an example value
mcr.cases{2,2} = 150;
mcr.cases{2,3} = 200;
% save your mcr file 
save('mcr','mcr');
```

You can also check out a similar example in the applications repository:  ./Multiple_Condition_Runs/RM3_MCROPT3_SeaState (https://github.com/WEC-Sim/WEC-Sim_Applications)

Please let me know if this solves your issue.

kmruehl, 2021-04-30:
Resolved with PR #570

dforbush2, 2021-05-05:
Hi @Pisdan96,

Did this resolve your issue?

Pisdan96, 2021-05-06:
yes, indeed. Thank you very much :)

closed 2021-05-06 by Pisdan96