WEC-Sim/WEC-Sim #565: BEMIO and hydrodynamics inputs

hossein-mehdipour, 2021-04-27:
I don't know if i should ask this here or not, but
I want to know when determining the hydrodynamic forces via BEMIO, what are BEMIO's inputs exactly? (should I just look up the WAMIT, NEMOH & AQWA's inputs?)
and more especifically, if parameters like wave height, wave period, PTO damping & Stiffness Coefficients or Depth of water influences that?
I have another question
what is default depth of water in the default OSWEC directory files (in BEMIO)?

nathanmtom, 2021-04-28:
@hossein-mehdipour Thank you for your question and I will do my best to answer your question. 

BEMIO requires the hydrodynamic output files from either WAMIT, NEMOH, AQWA, and CAPYTAINE. Examples for the output files needed from WAMIT, NEMOH & AQWA can be found in the `examples/BEMIO/` on the WEC-Sim repository. 

The two parameters you have listed that influence the output of BEMIO is the wave period(s) and the water depth. These are defined within your hydrodynamic solver and is read by BEMIO to generate the .h5 read by WEC-Sim. 

The water depth for the OSWEC example case is 10.9 m. If you look at the oswec.out file in the `WEC-Sim/examples/OSWEC/hydroData/oswec.out`.

Cheers, 
Nathan
National Renewable Energy Laboratory

hossein-mehdipour, 2021-05-01:
@nathanmtom Thank you for your answer, I had another question if you don't mind,

I opened the oswec.out file in the `WEC-Sim/examples/OSWEC/hydroData/oswec.out`, I noticed that there are a lot of different Wave periods which is used in that file, and there are numerous added mass & damping coefficient calculated for each wave period, I don't understand the theory behind that, I would really appreciate it if you could point me to where I can find more about the theory behind this, or if there is a short and simple answer, you wouldn't mind sharing with me.

My initial assumption was that BEMIO uses a single wave period to calculate hydrodynamic coefficients, but now I see it uses multiple **wave periods, wave numbers & wave headings (wave directions)** as inputs, which is really confusing for me.

nathanmtom, 2021-05-01:
@hossein-mehdipour Thank you for your response and glad you've been able to find more information on BEMIO and linear hydrodynamic coefficients. 

In short, the wave radiation and wave excitation coefficients are calculated in WAMIT, NEMOH, or AQWA in the frequency domain that assumes a certain oscillation and wave frequency. Because of the wave dispersion relation, waves generated at different frequencies generate different pressures about a floating body that will then alter the hydrodynamics. WEC-Sim takes the frequency domain hydrodynamics and converts then into their time domain equivalent through Cummins Equation. 

I'd recommend looking through material in this open course for additional information. 

https://ocw.mit.edu/courses/mechanical-engineering/2-20-marine-hydrodynamics-13-021-spring-2005/lecture-notes/lecture22.pdf

Cheers, 
Nathan
National Renewable Energy Laboratory

hossein-mehdipour, 2021-05-03:
@nathanmtom Thank you for your short explanation and the additional link,

Let me check if I got it correctly, so in summary BEMIO pre-calculates the hydrodynamic forces for different wave periods, so when we give it a wave periods in `wecSimInputFile.m` it doesn't need to rerun the BEMIO and generate a new `.h5` file.
I hope I'm not completely in the wrong here.

I want to ask another question if you don't mind,
I want to run WEC-Sim for an OSWEC for my academic project, I have the same water depth as the default OSWEC case (10.9 m),
but I have different Wave period & ..., . Do I need to rerun BEMIO or not? Or I just need to run WEC-Sim?

Thank you so much again for taking the time.

nathanmtom, 2021-05-03:
@hossein-mehdipour Yes, you are correct that BEMIO will pre-calculate the hydrodynamic forces for the vector of wave periods that were defined in the hydrodynamic solvers (WAMIT, NEMOH, AQWA). So as long as the wave period defined in `wecSimInputFile.m` is within the vector calculated by the hydrodynamic solver you will not need to rerun BEMIO. You only need to rerun BEMIO if you have a new hydrodynamic data or you are unhappy with the output from the default BEMIO settings. 

Per the above comment, as long as you are not changing the hydrodynamic coefficients of the WEC you will not need to rerun BEMIO. If you want to change the wave period, or other parameters, you will need to update the `wecSimInputFile.m` or any other supplementary files you might have. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

hossein-mehdipour, 2021-05-04:
@nathanmtom thank you very much for clarifying the matter, I think I got it now.

closed 2021-05-04 by hossein-mehdipour