WEC-Sim/WEC-Sim #566: Energy of irregular wave in time domain and frequency domain

mmartini1987, 2021-04-28:
Hello
I would like to know more on how time-histories of wave elevation are generated for an irregular wave.

I have been generating two wave-elevation time-histories with equal input data (type = 'irregular', H = 1.25m, T = 9 s, spectrumType = 'J'), measured the energy content of the wave elevation in time-domain, compared it to that in frequency domain and noticed a significant difference. 

For Parseval's theorem, the m0 spectral moment of the wave in frequency domain is equal to the variance of the signal in time domain.

Attached an example with two wave classes ('waves1' and 'waves2') for your reference. 

The m0 spectral moment of these waves, calculated through the Matlab command trapz(wave.w,wave.S) is the same in both cases and equal to 0.097275 m2. If I calculate the variance in time-domain (through the Matlab function 'var'), for 'waves1' I get 0.1090 m2 and for 'waves2' I get 0.0932 m2. That is, 'waves1' in time-domain has 12% more energy than its twin in frequency domain, while 'waves2' in time-domain has 4% less energy than its twin in frequency domain.

I become aware of this while re-running an MCR for PTO optimisation - the mean PTO power for 'waves2' was 15% less than for 'waves1', which really should not be the case.

PS Keeping or removing the ramp time does not seem to solve the issue.

Any hint on this?

Thanks
Michele

[WaveExample.zip](https://github.com/WEC-Sim/WEC-Sim/files/6392176/WaveExample.zip)

nathanmtom, 2021-04-28:
@mmartini1987 Thank you for your question and I will do my best to answer your question. 

For clarification, have you defined the following `waves.phaseSeed` in your `wecSimInputFile.m`? In order to generate a time history of the wave elevation, WECSim uses a random number generator to randomize the phase of the individual sinusoidal components to generate the irregular wave profile. However, the randomization of the phase will be different each time you run WECSim unless you define `waves.phaseSeed`. 

Based on your comments above, it is consistent that the m0 calculation is the same for the wave objects as they are defined by the same spectral form. If the wave random phase components are different between the two wave profiles then I would expect the `var` or `std` outputs of the two time histories to vary. 

Cheers, 
Nathan
National Renewable Energy Laboratory

mmartini1987, 2021-04-29:
@nathanmtom Many thanks for your quick and clear answer.

I did not define waves.phaseSeed in wecSimInputFile.m. I indeed understand that this generates two different random wave profiles which, on turn, will have a different variance in time-domain. My concern is related to how much the energy of the random generated wave is different from that of its spectrum in frequency domain. 

Would it be useful performing a check on the wave profile just after it is generated and, if its energy content is too far from that in frequency domain (say, difference above 5%), generate a new wave profile?

Thanks again and have a good day

Michele

nathanmtom, 2021-04-29:
@mmartini1987 I'm glad my first response helped clarify some of your questions. 

I need a little more clarification on your statement on `energy of the random generated wave`. The amplitude of the individual sinusoidal components is defined as `A(\omega)^{2} = (1/2)S(\omega)d\omega`. Thus the wave elevation profile is defined by the wave spectrum defined by the Jonswap spectrum. Normally our team checks the energy of the wave by performing an FFT to generate the numerical spectrum of the wave elevation time series and compare it against the theoretical waves spectrum. Is this the same procedure you have used when you mention the energy content is too far off from the frequency domain?

But, now if I revisit you initial question I believe you want m0 = 'var(waves.waveAmpTime)'? If so, theoretically the longer you extend the wave time history the closer that expression will hold regardless of the phase. If you only want to run a 2000 s or lower time history than there will be some greater variation and you could run a check on the wave profile after it is generated to see if the variance of the time signal is sufficiently close to your m0 calculation. 

Cheers, 
Nathan
National Renewable Energy Laboratory

nathanmtom, 2021-05-05:
@mmartini1987 I wanted to check with you to see if you had any further questions on this issue? If not please close this issue to indicate this is resolved. 

Cheers, 
Nathan
National Renewable Energy Laboratory

mmartini1987, 2021-05-06:
Hi Nathan
Apologies for the delay. I will run some trials on my own to get a better undestanding on the statistics of wave elevation time-series but I think there is no more need for support from your side. You can close the issue.

Many thanks for your prompt help.

Kind regards,
Michele

closed 2021-05-06 by nathanmtom