WEC-Sim/WEC-Sim #567: Analyzing Hydrodynamic Values

sugarl14, 2021-04-28:
Hiya! 

I talked with @dforbush2 a few months ago in Issue #480 about Hydrodynamic Coefficients that are acceptable. He mentioned that we should try to be looking to get the radiation damping and wave excitation terms to approach 0 as wave frequency approaches infinity. Likewise, added mass should approach a constant horizontal asymptote as wave freq approached infinity. It is my understanding that we do this to ensure the impulse response functions (IRFs) are totally stable, and they themselves approach 0 as frequency approaches infinity.

I'm looking to discuss this idea in my thesis and will some citations! Do y'all know of any publications that talk about this idea directly? So far, I'm thinking of using [this ](https://www.nrel.gov/docs/fy15osti/63528.pdf) publication from @nathanmtom, but wasn't sure if you all had any more suggestions!

Thank you! :)

Landon

nathanmtom, 2021-04-28:
@sugarl14 Thank you for your question. The reference you have cited does discuss the approach taken within WEC-Sim for developing the state space representation of the impulse response system. I would consider reviewing the references cited in that paper for more information from a theoretical standpoint. Specifically reference [14] Taghipour et. al. 2008 and [16] Kristiansen et. al 2006. Note only the wave radiation damping term is used within WECSim to calculate the impulse response functions and it is preferred to see that at the highest frequency in your BEM solution the corresponding value is near zero to satisfy the mathematical expression to derive the IRF.

sugarl14, 2021-04-29:
@nathanmtom Awesome, thank you for the help!!

closed 2021-04-29 by sugarl14