WEC-Sim/WEC-Sim #568: PTO block and contraints

ChrisJohnsonJ, 2021-04-29:
Hi, 

I am trying to model a simplified in-plane pendulum using WEC-Sim blocks. I have attached a picture of the case as follows:
[image]

 I had a few queries and are as follows:

1. The rotational PTO actuation force block constraints in Y direction.  Is there a way to constraint in different directions apart from Z and Y?
2. The Floating 3DOF constraint block, allows motion in Translation in x,y and rotation in z (Surge, Sway and Yaw). Is there a way I can constraint motion with translation in y and rotation about x and z(Sway, Roll and Yaw), so that I allow motion in Surge, Heave and Pitch?

I look forward to your response! Thanks in advance!

Chris

dforbush2, 2021-04-29:
Hi Chris, 

This is a cool application. 

With respect to constraints:

1). Yes, though by your diagram, the default rotational PTO should be what you need, since it allows rotation about an axis perpendicular to the wave propagation direction (X) and gravity (Z). You can use rotation matrices to allow rotation only about an arbitrary axis by defining `constraint(#).orientation.rotationMatrix` in your WEC-Sim input file. Within the block, it this is utilized: 

[image]

2). The Floating 3DOF constraint, by default, allows motion in surge, heave, and pitch. So you should be good to go here, but you can follow a similar approach to the above to define free axes if necessary. 

In case you end up needing to limit motion extent, I'd like to call your attention to the applications case ./End_Stops (https://github.com/WEC-Sim/WEC-Sim_Applications). There is a Matlab version dependency, so I recommend different workflows whether or not you are using v2019a or newer. This applications case covers both.

kmruehl, 2021-04-30:
Thanks @dforbush2 , this also relates to my response on #559

kmruehl, 2021-05-05:
@ChrisJohnsonJ  does this answer your question?

ChrisJohnsonJ, 2021-05-06:
@dforbush2 @kmruehl 

I apologize for the delayed response! Thank you so much for your suggestions! I thought I have to change the constraint block that comes after the rigid transform block! I was able to get some results but I may have a few questions regarding this! I will try to work on it and if I am stuck I will reach out to you! Thank you once again for your guidance!

closed 2021-05-12 by yuyihsiang