WEC-Sim/WEC-Sim #569: Small WEC model consistent singularity

t-osu, 2021-04-29:
Morning all,

I have been struggling with modelling a small WEC for a long time. It takes on the shape of a tapered cylinder at the bottom and top, and consistently causes a singularity in the WEC-Sim solution not even one second in. At the time I first posted about this, I was using AQWA, and was told to try a different technique with better meshing. This past week I Iearned Rhino & WAMIT and still have the same issue. I'm not an expert 'mesher' though, but I don't think that is the main issue. I then tried including non-linear hydrodynamics in the WEC-Sim solver as suggested by an advisor with no luck. In previous posts, a WEC-Sim developer took a look at the output of the _bemio.m_ files and said they looked good. I got similar bemio plot results with the Rhino & WAMIT code and also can see that, at least the added mass and damping, begin to converge to a single value.

The WEC specifically is 0.7 m in diameter and 0.32 m high, included in the image below. I would really like to get this model running to use it as validation for tank testing I am doing in a few months, so I have set the water depth as ~1.5 m in my WAMIT runs. The prototype WEC is actually already built from a previous students work, so the dimensions and mass properties cannot really be changed that much. 

I took a quick look at some force values and from what I can see the added mass is REALLY high, but I don't have the proper background in hydrodynamics to debug this, or if it is even the issue.

Any guidance is appreciated,

- Chris 
[image]

akeeste, 2021-04-30:
Hi @t-osu ,

I will do my best to help you with your singularity issue. Before I can do that, will you upload a zip file of your most recent case directory and tag any previous GitHub issues with this device for my reference?

As a start, I recommend doing the following cases in order (this will also be part of my own workflow to debug your case):
1. noWave and noWaveCIC. This tests for numerical hydrostatic stability of the given displaced volume/mass.
2. noWave and noWaveCIC + initial displacement. This will test for hydrostatic stability and test the hydrostatic stiffness of the device.
3. regular and regularCIC waves. This will begin testing the influence of the BEM coefficients for a realistic response. 
4. irregular wave. 

If you have not gone through these steps, I highly recommend doing so. Stable solutions for the preceding steps are essential for the next cases to succeed. For example you might get a regular wave case to work, but it will not be robust if all cases in steps 1 and 2 are not stable. If you have gone through this workflow, please let me know which step you are finding issues at.

We are working to add a documentation example of working through this process for a new WEC model. In the meantime, if you have suggestions on content for this workflow or getting started advice please let me know!

Adam

t-osu, 2021-05-03:
Hi @akeeste,

For _noWave_ I'm unsure where to find the period of BEM data used for hydrodynamic coefficients because I had multiple periods simulated in WAMIT, so I just took the upper end of my range (but correct me if wrong). Either way, both _noWave_ and _noWaveCIC_ worked only when the mass was set to equilibrium. It should be around 24 kg, but with equilibrium it is 29.4669 kg; guess that is not too terribly off.

I tried multiple linear initial displacements (0.01, 0.1, and 1 m ) and these all failed, so this appears to be where the issue is.
Looking forward to the next steps!

- Chris

akeeste, 2021-05-03:
Hi @t-osu 

If the simulation is failing with the noWave case, the device is not hydrostatically stable. From what you have said, it seems the device is too light for the given submerged volume in your BEM data. Its likely the buoyant force is larger than the gravitational force, and will force the body to continuously rise until the simulation crashes.

There are a few ways to address this: 
1. add mass (for example increase to equilibrium mass)
2. rerun the BEM solution with the body at a higher vertical position, decreasing the submerged volume and buoyant force.
3. manually tune the hydrostatic stiffness. 
4. use a mooring and pretension to hold the device down and prevent it rising indefinitely (if appropriate for your system).

Since you already have a built prototype, I would use options 2 and 3. Calculate the device's equilibrium position relative to the SWL based on the mass and geometry. Compare this to the current location in WAMIT. Setting this correctly allows you to start at the equilibrium position in WEC-Sim. After confirming/fixing that, try tuning the hydrostatic stiffness. This is the quantity that will force the device to its equilibrium position when given an initial displacement. Users can overwrite the BEM input and tune this by defining body(#).hydroStiffness in the wecSimInputFile (6x6 matrix). If the stiffness is too low, it will not prevent the device from indefinitely rising and crashing the simulation. If the stiffness is extremely high, it can cause the device to oscillate unphysically and crash.

If you are okay sharing your case files, please upload them here and I can take a closer look for other issues that may be causing this singularity problem.

Adam

t-osu, 2021-05-04:
Not a problem, attached are the files. I appreciate the look-over.
I'll give those ideas a try tomorrow and report back what I find! Thanks again.

[VPWEC.zip](https://github.com/WEC-Sim/WEC-Sim/files/6418811/VPWEC.zip)

akeeste, 2021-05-04:
Hi @t-osu 

What WEC-Sim version are you using? It seems that your model is not working because you are not using the latest version, or have made an incorrect change in the hydrodynamic body. 

The Simulink library link on your hydrodynamic body is broken and is not running the current version of WEC-Sim. The hydro body appears to be using v4.1 or previous. If the rest of the repo is at a different version, it may cause you some issues.

I restored the link and ran the case just fine with noWave, noWaveCIC, regular waves using the current version of the master branch.

Adam

t-osu, 2021-05-04:
Yes, restoring the link causes the simulation to work, that must have been an old overlooked edit, but knowing the above debug process is very helpful for future simulations. Albeit annoyed at myself, glad it was an easy fix!

I do have another question though. I am trying to attach mooring to the model using MoorDyn which I have successfully used before, however this time I am encountering an error I have never seen:
Error: NaN value detected in MoorDyn state at dynamic relaxation time 1 s.

I looked it up and Matt Hall helped another user having an issue [(Matt Hall answer)](https://wind.nrel.gov/forum/wind/viewtopic.php?p=13187). I tried the suggestions (node number, depth and vessel location), but neither worked. The version of MoorDyn is also the one he suggested, 1.01.00.
Thoughts on that?

kmruehl, 2021-05-05:
@t-osu  thanks for the update, can we close this issue?

t-osu, 2021-05-06:
Yes, I unfortunately have a separate issue after trying to update my files with git pull that I will start a new issue on after a long day of attempted debugging.
Thank you both

closed 2021-05-06 by t-osu