WEC-Sim/WEC-Sim #572: Use WEC-sim of a WEC attached to the shore?

Nchals, 2021-04-30:
Hi,
I am really new to WEC-sim and have just started to learn a bit about your creation and think it's such a nice tool! Currently I'm in a school project where we've developed a WEC, however, I have a question regarding if you think it would be possible to create and simulate a point absorber WEC attached to the shore through WEC-sim? Or do you think it would be better for us to use another program to simulate the WEC? (We are two engineer-students working on this with only 2-3 weeks left on our project)

Kind regards
Nelly

nathanmtom, 2021-05-01:
@Nchals Thank you for your question and will do my best to answer. 

Yes, WEC-Sim has been used to model a point absorber attached to the seabed. Generally this has been done by connecting your WEC hydrodynamic body to the Global Reference Frame using a translational PTO. This does constrain the WEC to only oscillate in heave which may or may not be a good representation of the system you want to model. 

My suggestion is to review the `Free Decay` example we have in the WEC-Sim applications folder. 

https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Free_Decay

Let me know if this has answered your question.

Cheers, 
Nathan
National Renewable Energy Laboratory

Nchals, 2021-05-05:
Thank you for your fast answer!
But have the WEC-sim been used to model a point absorber attached to the shore?
Kind regards,
Nelly

nathanmtom, 2021-05-05:
@Nchals Thank you for your clarification and reference to Eco Wave Power since it helps better understand your direct need. 

WEC-Sim can model the [Eco Wave Power](https://www.ecowavepower.com/) concept although the development team does not have an applications case for reference However, the [WECCCOMP](https://github.com/WEC-Sim/WECCCOMP) repository modeled a [WaveStar ](http://wavestarenergy.com/) arm which is very similar to the hinged motion like Eco Wave Power. The main difference would be that Eco Wave Power appears to be placed off of a breakwater or stone pier so behind it is a wall that will reflect incident waves. This could be captured by modeling a wall behind the float in your hydrodynamic model. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

nathanmtom, 2021-05-12:
@Nchals I wanted to check-in to see if you had any additional questions about modeling your desired WEC within WEC-Sim? If we've answered your question, we request that you please close this issue. 

Cheers, 
Nathan
National Renewable Energy Laboratory

Nchals, 2021-05-15:
Hi,
Thank you very much for your response! It has been very instructive and helpful having the Wecccomp as a reference.

Although I may have one additional question. Below I attach a screenshot of when I've been trying to simulate my version through WEC-sim. It appears I have multiple errors and also I can't see the WEC in the "Mechanics explorer"-window. Is it possible for you to discover/distinguish some of the things I have to correct? Or help me in the right direction?

[image]

nathanmtom, 2021-05-17:
@Nchals Thank you for the update on your model development and glad the previous information helped in your own work. 

To your questions: 

1. In the `Mechanics explorer` I am assuming that you have generated your own `.stl` file for visualization correct? If so, I would double check that the units of your `.stl` file is in meters.
2. In response to your error, generally that error occurs when we see two joints connected in series as at times we need a dummy mass to allow Simscape to have a non-zero mass in the calculation of the dynamics of joints or constraints. It sounds like you have modified the base WECCCOMP Simulink model to fit your needs and will need to know more about those changes before I can provide further reccomendations. 

Cheers, 
Nathan
National Renewable Energy Laboratory

Nchals, 2021-05-17:
Thank you for all help!!
I did manage to solve the simulations earlier today.

Kind regards,
Nelly

closed 2021-05-17 by Nchals