WEC-Sim/WEC-Sim #573: What is the relation between inertia force and others in wec-sim?

zzq1019, 2021-05-01:
In the simulation of RM3 project, I want to calculate the inertia force: ma. the simu.b2b = 0. output.bodies(1). From my understanding:
M = diag([body(1,1).mass, body(1,1).mass, body(1,1).mass, body(1,1).momOfInertia])
a = output.bodies(1).acceleration';
Ma = output.bodies(1).forceTotal + output.ptos.forctToal
At t=0, the ptoForce = 0, so the ma of body1 is equal to the forceTotal of body1.
At least in z-axis, I think the above formulations are correct. However, the fact is not.
So, what is the real relation between inertia force and others?
I look forward to your response. Thanks in advance!
[image]
[image]

[image]
[image]

[image]
[image]

kmruehl, 2021-05-05:
@zzq1019  Can you please clarify this request? Are you trying to calculate the inertial force as F=ma?

zzq1019, 2021-05-06:
Thanks @kmruehl for the reply.
Sorry for my poor english. And, yes, I am trying tio calculate the inertial force using the simulation results:`output`
The equation of motion for a floating body can be given as:
[image]

In the simulation of RM3, I want to verify the simmulation results about the equation above. The `wecSimInputFile.m` as:
[image]
[image]

From my understanding, for the body(1):
`F_ext + F_rad + F_B = output.bodies(1).forceExcitation - output.bodies(1).forceRadiationDamping -output.bodies(1).forceAddedMass - forceRestoring = output.bodies(1).foreTotal`
`F_v = 0, F_me = 0, F_m = 0`
`F_pto = output.bodies(1).forceTotal`

From other issuses, I have learned that all the forces act on the center of the object. So, in addition to the Z-axis motion, the above equation is not valid (I don't know if my understanding is correct). 
At least in Z-axis of body(1), it bears only two forces: the `output.bodies(1).forceTotal` of z-axis and `output.ptos.forceTotal` of z-axis. So, the equation of motion for body(1) shoul as: 
`body(1,1).mass * output.bodies(1).acceleration(:,3) = output.bodies(1).forceTotal(:,3) + output.ptos.forceTotal(:,3)`

In particular at time `t = 0`,
`output.ptos.forceTotal(:,3) = 0`
 `body(1,1).mass * output.bodies(1).acceleration(:,3) = output.bodies(1).forceTotal(:,3)`

In the simulation results, the above equation does not hold.
Is there anything obvious I might be missing here? 
Thanks

yuyihsiang, 2021-05-14:
Were you running the RM3 with 3DOFs? If so, I think the formulation you have won't held because output.ptos.forceTotal(:,3) is actually the translational force between the two bodies, not the heave force. You may want to test the same formulation by constraining the RM3 motion to heave only (i.e., replace the 3DOF block with a translational block) to test it out. We are definitly interested in the outcome.

Thanks,

--Yi-Hsiang

zzq1019, 2021-05-15:
Hi @yuyihsiang 
I know what you mean, the direction of `output.ptos.forceTotal(:,3)` is  along the center of mass of body(1) and body(2). And I have run the RM3 with translational block.
[image]

And we can see the position of body(1), it is only moving in the z direction.
[image]

The results are shown in the figures. 
 
[image]
[image]
[image]

`body(1,1).mass = 725834`
`body(1,1).mass * output.bodies(1).acceleration(:,3)`:
[image]

`output.bodies(1).forceTotal(:,3) + output.ptos.forceTotal(:,3)`
[image]

So, we can see that `body(1,1).mass * output.bodies(1).acceleration(:,3)` is not equal `output.bodies(1).forceTotal(:,3) + output.ptos.forceTotal(:,3)`

Thanks

yuyihsiang, 2021-05-26:
Thank you for testing it. I also looked into this and got something similar to what you have shown here. 

If you plot the two forces in time for several seconds, you will see a small difference between the two. The difference we see here was caused by the use of simu.adjMassWeightFun and the lag of added-mass term in time on the right-hand side of the equation. Therefore, the output.bodies(1).forceTotal(:,3) we see here is not the exact total force solved in the equation of motion. I did a test by zeroing out the added-mass term in surge and sway to zero and adjusting the simu.adjMassWeightFun to 1, which forces the infinite frequency added mass scalar value equal to the heave added mass value, and the two forces become identical.

--Yi-Hsiang

zzq1019, 2021-06-02:
Thanks for your reply @yuyihsiang

I have plotted the two force, and the difference between `ptos.forceTotal + body.forceTotal` and `m*body.acceleration` is small as you said. The slight difference will not affect my application.

closed 2021-06-02 by zzq1019