WEC-Sim/WEC-Sim #574: Hard stops on OSWEC example

mmartini1987, 2021-05-03:
Hello
I am running the OSWEC example and trying to include hard-stops in the modelling. I have modified the wecSimInputFile.m adding the following rows:

pto(1).hardStops.upperLimitSpecify = 'on'; % 
pto(1).hardStops.upperLimitBound = 5;  % [deg] 
pto(1).hardStops.upperLimitStiffness = 0; % [Nm/deg]
pto(1).hardStops.upperLimitDamping = 1e8; % [Nm/(deg/s)]
pto(1).hardStops.upperLimitTransitionRegionWidth = 0.1; % [deg] 

pto(1).hardStops.lowerLimitSpecify = pto(1).hardStops.upperLimitSpecify;
pto(1).hardStops.lowerLimitBound = -pto(1).hardStops.upperLimitBound;
pto(1).hardStops.lowerLimitStiffness = pto(1).hardStops.upperLimitStiffness;
pto(1).hardStops.lowerLimitDamping = pto(1).hardStops.upperLimitDamping;
pto(1).hardStops.lowerLimitTransitionRegionWidth = pto(1).hardStops.upperLimitTransitionRegionWidth;

This however seems not to affect the flap motion. To see any hard-stop effect occurring, I had to navigate into OSWEC.slx->Rotational PTO->Pitch PCC, force the "Specify Upper Limit" and "Specify Lower Limit" to 1, save the model (neglecting the warnings related to having modified a WEC-Sim library block) and run it again.

Are you able to replicate this - or maybe am I mistaking something?

Many thanks
Michele

dforbush2, 2021-05-03:
Hi Michele,

What version of Matlab are you running? It doesn't sound like the mask initialization function is running properly on the pto's in the OSWEC model. I am able to replicate your error in 2020b...though. 

For whatever reason, the try/catch statement in the mask initialization is not running because it does not recognize the block name that sets the pto(1).hardStops.upperLimitSpecify and pto(1).hardStops.lowerLimitSpecify toggle switches. So you don't get an error, but your limits don't get set. Only problem is these block names are observably identical to what I have coded there, so there is something subtle happening. I will dig into this and push a fix.

In the mean time, your work-around will work. Just make sure that you do not push the changes made locally to your wec sim library or that may break some of your other models!

Best,
Dom

dforbush2, 2021-05-03:
Got it. There is a newline character that only is visible in some ways of viewing the block name that exists in the simulink model but did not exist in the mask code. Fix imminent.

mmartini1987, 2021-05-03:
Hi Dom
Many thanks for this, very useful. Looking forward to download the update.

Kind regards,
Michele.

mmartini1987, 2021-05-18:
Hi,
I updated WEC-Sim through the "git pull" command, but now I have this error popping up for all my models:

Error using simulationClass/loadSimMechModel (line 106)
block_diagram does not have a parameter named 'SimMechanicsOpenEditorOnUpdate'

Error in wecSim (line 375)
simu.loadSimMechModel(simu.simMechanicsFile); - Show complete stack trace

Can you please give me some support?

Michele

dforbush2, 2021-05-18:
Hi Michele,

To avoid confusion in the future I'm going to close this issue (related to hard stops, resolved by PR #574)  and open a new one. This pull request has not yet been merged into the master library so it is not the source of this issue.

Best,
Dom

closed 2021-05-18 by dforbush2