WEC-Sim/WEC-Sim #576: 2021a matlab and WEC-SIM

nimmy91, 2021-05-04:
I have been trying to install the latest version of WEC-SIM in my laptop as well as department system, which are both of Matlab 2021a version. I found an update regarding the error fixing in 2021a version 25 days ago and I reinstalled everything which went fine. But after a day, when I came back for WEC-SIM, it's missing from my path as it was from the updated time. Everytime I install the same and when I comeback after a reboot, it will vanish. I don't know what I am supposed to do now. Could you please help me to fix this issue.

akeeste, 2021-05-04:
Hi @nimmy91 

As I understand it, you have installed WEC-Sim correctly, but when you restart MATLAB "/WEC-Sim/source/" is not on the Matlab path?

Please follow the instructions on our [Getting Started documentation](https://wec-sim.github.io/WEC-Sim/man/getting_started.html#step-1-add-wec-sim-to-matlab-path). This page describes how to add the WEC-Sim path in the Matlab startup.m file so that you can use WEC-Sim immediately upon restarting Matlab.

Let me know if this resolves your issue.

Adam

nimmy91, 2021-05-05:
Dear Adam Keester, 
I have done the same steps before during the installation.
Warm regards,Nimmy Thomas

akeeste, 2021-05-05:
Hi @nimmy91 

Can you please give more details on your issue? Please copy/paste your error message and upload your startup file to assist us in debugging this issue.

Thanks
Adam

akeeste, 2021-05-17:
@nimmy91 I will close this issue due to a lack of interaction. If you still have the same problems please reopen this issue or a new one.

Thanks

closed 2021-05-17 by akeeste