WEC-Sim/WEC-Sim #577: Hydrostatic Restoring Forcing Function

sugarl14, 2021-05-05:
Hi, 

I am looking to create a function that calculates the hydrostatic restoring force of an OWSC at any given theta angle. I know the hydrostatic coefficient is given by C = rho*g*V, where V = the volume of displaced fluid. 

Just by looking at an FBD, I would think that the remainder of the function would be cos(theta*t), where theta is the angle, and t is time, so combining would yield:

Hydrostatic restoring forcing function = rho*g*V * cos(theta*t) 

I'm not sure about my answer, so I was wondering if you all knew of any documentation  (maybe a journal article, or somewhere in the wec-sim code) that could help me determine this?

I'm looking to use it in a user-defined-function for ANSYS Fluent so I can prescribe a restoring force to the OWSC after it's hit with a wave. :)

thank you!

Landon

nathanmtom, 2021-05-05:
@sugarl14 Thanks for your question and I will do my best to answer your question. 

To your first point the vertical restoring force in the vertical direction is given by the equation you have written, but that expression is not complete if you are looking to calculate the instantaneous restoring moment about the rotation point of an OSWEC. Note I am having some confusion as the expression you have listed has units of force and not sure if you are intending it to be a moment which would be force*distance. 

Do you have a schematic where you are defining the angle \theta or can attached your free body diagram? Normally a \theta = 0 correspond to the OSWEC oriented vertically at which point your expression would be maximum and if the OSWEC were to move to be horizontal there would be no restoring force which should not be the case unless I am misunderstanding the set-up of your problem. 

The nonlinear rotational hydrostatic moment for an OSWEC is generally described by: 

[image]

where \zeta_{5}(t) is the rotation angle about the bottom hinge which I've assumed is the same as \theta(t). 

Before I can suggest some other documentation, I believe we need some additional information from you to clarify my questions above.

Cheers, 
Nathan
National Renewable Energy Laboratory

sugarl14, 2021-05-05:
@nathanmtom Thank you for your response. Sorry to cause confusion-- yes I am looking for to calculate the instantaneous restoring moment about the OWSC's pivot axis (denoted by P in the FBD below). I used \zeta){5}(t) and \theta interchangeably here. But from hereon, I will refer to the angle change as \zeta){5}(t).

[image]

nathanmtom, 2021-05-05:
@sugarl14 Thank you for that clarification. Then the expression listed in my first comment would be the nonlinear expression that describes the instantaneous hydrostatic restoring moment for the OSWEC. 

WEC-Sim currently uses the linearized hydrostatic restoring coefficients to calculate the response and does not have a default option to implement that expression. The linearization of the pitch hydrostatic moment is appropriate up to a rotation angle of around 30 degrees. If you desire to model the nonlinear equation in WEC-Sim, then you could edit the hydrostatic restoring block within WEC-Sim to output the desired forcing as all of the information listed in the expression is available in that block (or can be). 

Another reference to consider can be found **[HERE](https://www.sciencedirect.com/science/article/pii/S0029801814004818)**.

Cheers, 
Nathan 
National Renewable Energy Laboratory

kmruehl, 2021-05-05:
@sugarl14 [here is another reference](https://www.sciencedirect.com/science/article/pii/S0029801818308151) on this topic.

sugarl14, 2021-05-06:
Thank you, both, for your help!!

closed 2021-05-06 by sugarl14

hossein-mehdipour, 2021-05-25:
@nathanmtom 
Excuse me for bothering you but I had a question about something you said in this thread:
"The linearization of the pitch hydrostatic moment is appropriate up to a rotation angle of around 30 degrees"
I've seen this sentence in multiple papers that "When using the Linear Water Wave Theory, the Excursions of the flaps should be small." , but unfortunately none of them explained what is the specific reasoning behind this assumption, which equation (if any) is changed from non-linear to linear theory using this assumption?

Thank you!

nathanmtom, 2021-05-25:
@hossein-mehdipour Thanks for your question and happy to try and answer. 

1. The full equation for the pitch hydrostatic moment is nonlinear, where the moments between the center of gravity and the center of buoyancy are multiplied by \sin(\theta) where \theta is the angle of rotation from it's equilibrium condition which is generally vertical. \sin is a nonlinear function, but if the rotation is within 30 degrees then \sin(\theta) \approx \theta so the expression can be linearized. 
2. The second is that when using linear water wave theory to calculate the hydrodynamics, the assumption is that the body makes small oscillations about the equilibrium position. If the submerged or floating body begins to make large amplitude motion the theory begins to break down. To this point there are a lot of other nonlinear forcing that could be important but will be dependent on the specific shape and operation. 

I hope this answers your question but feel free to ask for clarification. 

Cheers, 
Nathan
National Renewable Energy Laboratory