WEC-Sim/WEC-Sim #578: Power Optimization (latching / declutching)

Pisdan96, 2021-05-05:
Hi again :)
I am trying to implement a latching control to a point absorber model of my own and I went through [https://libstore.ugent.be/fulltxt/RUG01/002/494/968/RUG01-002494968_2018_0001_AC.pdf](url) and there is a method about applying a latching control in WecSim. 
However, I could not find any example or application in the repository.
I'd be grateful if anyone could help on this matter. 

Best regards

nathanmtom, 2021-05-05:
@Pisdan96 That is correct that you can implement latching control within WEC-Sim to follow the procedure outlined in the attached reference. However, this publication was done outside of the WEC-Sim development team so we do not have an example or applications case that implements latching. Perhaps you can see if the author of that reference has their own GitHub or code repository where they are sharing their WEC-Sim model with the community.

Cheers, 
Nathan
National Renewable Energy Laboratory

closed 2021-05-05 by nathanmtom

Pisdan96, 2021-05-05:
Is there any optimization for PTO control developed by the wec-sim team?

nathanmtom, 2021-05-05:
@Pisdan96 I'll need more clarification about the type of optimization for PTO control you are asking about. Unfortunately, because there is a wide range of WEC and PTO designs it is rather difficult to develop a single optimization process to suit every case. Generally, the WEC-Sim team uses our multiple condition runs (MCR) options to cycle through various PTO parameters to find those parameters that provide the desired performance. In terms of latching, we definitely do not have any tools available within WEC-Sim that would allow you to optimize or implement said control strategy.

Cheers, 
Nathan
National Renewable Energy Laboratory

Pisdan96, 2021-05-06:
@nathanmtom  Thank you. your reply was very clear.

I am finding a way to implement a system that when the (velocity >0), the force is captured by the given damping value and when the velocity is equal or near to zero, force capture is stopped. (more clear on the attached picture)
[image]

in order to do that, where should I change the code? (fyi I'm a newbie to simulink ).
My wec is a single body spherical heaving point absorber

Best regards
Pisindu Danusha

nathanmtom, 2021-05-06:
@Pisdan96 In WEC-Sim there are PTO plots that will take as input a force or torque that is calculated by the user. In this case you can create a Simulink sub-block that takes inputs, such as velocity, to set up and `if` statement to implement the control structure in the figure you have included. 

I'll point you to our [WECCCOMP application](https://github.com/WEC-Sim/WECCCOMP) which provides an example of how a user can implement their own custom controller within the WEC-Sim framework. 

If you have any other questions please open a new issue as the conversation has drifted from the topic you submitted in your initial issue request. 

Cheers, 
Nathan
National Renewable Energy Laboratory

hossein-mehdipour, 2021-06-08:
Hello @nathanmtom 
Excuse me for bothering you, but I have a question about Hydraulic PTO and control strategies, If you don't mind giving me a little more insight on the subject matter,

I have searched the literature and I couldn't find a proper study about and Hydraulic PTO unit parameters optimization (except for 2 studies ([1](https://iopscience.iop.org/article/10.1088/1755-1315/463/1/012129/meta) & [2](https://www.mdpi.com/1996-1073/14/1/79)) which are basically the same about a rather unconventional WEC), and there is a lot of studies about optimization of control strategies of the WECs. 

My question is, is it wrong/ illogical/ impractical to optimize different parameters of an HPTO for a WEC (like an OSWEC or a Heaving point Absorber) or it's not but it just happens that noone (to my knowledge) did that.

Cheers,
Hossein

nathanmtom, 2021-06-09:
@hossein-mehdipour Thanks for the question and I'll provide my thoughts although I am not an expert in hydraulic PTOs. 

To your question, in my opinion it is not impractical to optimize the various parameters of a HPTO for a WEC; however, based on working with PTO-Sim there are quite a few variables that one would need to optimize over and these parameters could change with sea state so the size of the problem can become quite large quite fast. I'm not familiar with how the other references were simulated, but our PTO-Sim does provide a higher fidelity model, but the internal solver for the hydraulic component can significantly slow down computational speed so it will take longer and you may need to wait hours (or days) for a single optimization cycle to run. 

To start out you may want to begin with more bounds or fewer optimization variables to see if performing a larger optimization project makes sense. 

I hope this has answered your question, but let me know if you have any other comments. However, I ask that if we moving further away from the topic of this issue #578 that you open a new issue. 

Cheers, 
Nathan
National Renewable Energy Laboratory