WEC-Sim/WEC-Sim #579: Input files for wec-sim

Swarooprajashekar, 2021-05-05:
Hi, we are group of 6 students working on project Wave Generator for remote island. No student has expertise in wave, but it was out our interest. But we designed physical model but to simulate in WEC-sim we have no idea. We figured out to add some input files but we are stuck in creating a h5 file for hydrodynamic data. How to create a h5 file for the given model?

nathanmtom, 2021-05-05:
@Swarooprajashekar Thank you for your question and we do our best to try and answer your question. 

If I have understood your question correctly, you are looking for assistance in generating a .h5 file for your model? 

If so then to clarify, WEC-Sim is not a hydrodynamic solver and uses the outputs from said solvers (WAMIT, NEMOH, AQWA, etc.) to calculate the dynamics of ocean bodies. 

To help, the WEC-Sim team has developed another repository **[HERE](https://github.com/WEC-Sim/MECC)** that provides some examples on how to use NEMOH and Capytaine which are open source hydrodynamic solvers. After you run NEMOH or Capytaine you will then run BEMIO.m within WEC-Sim to generate the .h5 file.

Cheers, 
Nathan 
National Renewable Energy Laboratory

nathanmtom, 2021-05-12:
@Swarooprajashekar I wanted to check with you to see if you had any additional questions about the topic you've discussed in this issue? If not, then could you please close this issue. 

Cheers, 
Nathan
National Renewable Energy Laboratory

Swarooprajashekar, 2021-05-13:
hi nathan thank you so much, it was really helpful.

closed 2021-05-13 by Swarooprajashekar