WEC-Sim/WEC-Sim #581: Simulation issues after a git pull

t-osu, 2021-05-06:
Hi all,
I wanted to make sure that my files were up to date so I used git pull through GitBash in my WEC-Sim folder. As a note, before the pull my model was working great. The git pull seemed to work, however afterwards I ran my simulation, but it failed. The behavior of the model was very different than before. I also noticed that it gave out warnings with my non-hydro bodies that the center of buoyancy was not defined and was then set to the center of gravity. Before the pull with my non-hydro bodies, I had never specified the center of buoyancy and a warning never showed up. I guess that is part of the newer version I downloaded after looking at the source file.
I also noticed that my library blocks had changed and do not show the link icon anymore or the option to go under into the block and view its subsystems. 
I tried removing everything and starting from scratch cloning the WEC-Sim files through GitBash, recreating the Simulink model, and am still seeing the same problems.

I attempted downloading the previous release, 4.1, and then got an error related to morisonElement (was about not enough inputs) which I do not have set and was also not having an issue with before. I tried setting it equal to zero just to see, but that didn't help.

I'm unsure why these are suddenly appearing, and would appreciate any guidance other than what I have thought of so far, or if I've stepped over something obvious.
Many thanks..

kmruehl, 2021-05-06:
@t-osu Thanks for letting us know! We'll look into this issue immediately as it sounds like a bug with the WEC-Sim Library. Just to verify, did you pull from the [master branch](https://github.com/WEC-Sim/WEC-Sim/tree/master), or the [dev branch](https://github.com/WEC-Sim/WEC-Sim/tree/dev)?

t-osu, 2021-05-06:
I'm unsure, I was following the general directions in the Getting Started page on the main WEC-Sim website and when it said to update using git pull I assumed in GitBash I just navigate to my WEC-Sim folder and then use the git pull command. Would that pull from the development instead of the master branch?

dforbush2, 2021-05-11:
Hi @t-osu 

What version of matlab are you using? My fresh pull also gives errors but only on 2021a.

kmruehl, 2021-05-19:
@t-osu can you provide a screenshot of your error and upload the WEC-Sim case?

t-osu, 2021-05-21:
Sure, I attached the error I was getting with the _wecSim.m_ I was using after the gitpull. I got around this by just using an older WEC-Sim source folder that my friend had from a previous download. 
I have since downloaded the zip file of the most up to date version (this morning, actually) and everything is working, giving out the same outputs as my friends version. I think whatever the issue was has been fixed, then?
Sorry, I keep trying to upload the case as a ZIP but am getting a "is not included in the list" error. I've tried changing the name with no resolve. I can upload other ZIP files, however this particular one is not working. If you want I can upload it to a google drive and give you access?

[image]

t-osu, 2021-05-21:
Sure, I attached the error I was getting with the wecSim.m I was using after the gitpull. I got around this by just using an older WEC-Sim source folder that my friend had from a previous download.
I have since downloaded the zip file of the most up to date version (this morning, actually) and everything is working, giving out the same outputs as my friends version. I think whatever the issue was has been fixed, then?
Sorry, I keep trying to upload the case as a ZIP but am getting a "is not included in the list" error. I've tried changing the name with no resolve. I can upload other ZIP files, however this particular one is not working. If you want I can upload it to a google drive and give you access?
[image]

kmruehl, 2021-05-26:
@t-osu I'm glad that resolved your issue. As you mentioned, this issue should have been resolved from a prior commit, so I think you were actually having issues with your `git pull`. I recommend re-cloning the WEC-Sim repository with a fresh `git clone`, according to the [WEC-Sim download instructions](http://wec-sim.github.io/WEC-Sim/man/getting_started.html#download-wec-sim). Alternatively you can manually download the ZIP, as you've done, but this should allow you to get the latest version of WEC-Sim via git. Thanks!

closed 2021-05-26 by kmruehl