WEC-Sim/WEC-Sim #585: Review implementation of convolution integral in radiation problem

johannes-spinneken, 2021-05-12:
[image]

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.** 

**Is your feature request related to a problem? Please describe.**
We compared the execution time of the convolution vs. the state-space implementation within the radiation problem. The attached image shows the comparison. For context, we ran WECSim on a Speedgoat Real-Time system using Simulink Real-Time (compiled C code generated from Simulink). While we would expect some difference between the convolution and state-space approaches, the execution time difference observed was larger than we expected. In the attached image, Baseline refers to a Baseline Speedgoat system, and Performance relates to a Performance Speedgoat system. The convolution time window was 20s.

**Describe the solution you'd like**
Review the implementation of the convolution implementation. Can the execution time of the convolution operation be improved to be closer to that of the state-space implementation?

**Describe alternatives you've considered**
We can run the state-space radiation implementation to meet our performance targets (e.g. computing a time step in <0.001s on our real-time target), but this requires additional steps (formulation of the state-space system) which may introduce inaccuracies in the radiation problem. 

**Personal project context**
Running WECSim in real-time for the FOSWEC device, such that it can be used for HIL work.

**Additional context**
N/A

yuyihsiang, 2021-05-13:
The convolution integration calculation (CIC) time really depends on your convolution time window AND time step size. If your time step size is small, e.g., 0.001s, the execution time of the convolution will be much longer than the execution time for the state-space model. In my experience, WEC-Sim simulation is about real time or slightly faster than real time on my laptop when using convolution integration calculation with a time step size of 0.01 s and a convolution time window of 30 s. In addition, the CIC was coded in MATLAB Function with some additional computation overhead, where we have to use the "persistent" variable in the function. I would expect it significantly slows down the simulation. 

Thank you for bringing up the question. We will take a look and see if it is possible to refine the code and make it faster. 

--Yi-Hsiang

kmruehl, 2021-05-26:
@johannes-spinneken we've added this to our list of priorities, but we are curious is there is a _need_ for the CI formulation using HIL, or whether SS is sufficient. Thanks!

johannes-spinneken, 2021-06-10:
> @johannes-spinneken we've added this to our list of priorities, but we are curious is there is a _need_ for the CIC formulation using HIL, or whether SS is sufficient. Thanks!

@kmruehl - apologies for the delayed response. I feel it would be better to have this, as the direct convolution avoids additional steps in generating the SS model. We are working on a project that would very much benefit from an improved direct convolution performance for HIL.

kmruehl, 2022-08-12:
Re-opening the issue, the SS formulation is not ideal for experimentally derived radiation coefficients (where you don't want a SS fit). This relates to the https://github.com/PMEC-OSU/FOSTWIN, and support of real-time simulations like Digital Twins.

ryancoe, 2022-08-12:
@johannes-spinneken @gbacelli - Can either of you recall what our proposed solution was? I think it had something to do with an interpolation?

johannes-spinneken, 2022-08-12:
@ryancoe @kmruehl Looking at this code:

https://github.com/WEC-Sim/WEC-Sim/blob/7adaaa8407e7f0a415315ce4df21ce62f4827851/source/functions/simulink/model/ConvolutionIntegral_interp.m#L27-L48

Some thoughts:
- interp_factor is set to 1 - I guess this could be increased to make the convolution integral faster? Given that both the IRF and the velocity signal are likely continuous and slowly changing, there is probably little difference in taking the integral over every timestep vs. say using every 10 time steps.
- pretty much any real-time Simulink implementation will use a fixed step solver. For a fixed time step, the trapz function could be replaced by sum()*Ts, see example below

```Matlab
% compare trapz vs sum

clearvars; close all; clc

Ts = 1/1000;

t = 0:Ts:60.5;

signal = sin(2*pi*1*t);

tic
result1 = trapz(t,signal);
toc

tic
result2 = sum(signal)*Ts;
toc

fprintf('result 1 = %6.5f\n',result1)
fprintf('result 2 = %6.5f\n',result2)

```

which gives
```bash
Elapsed time is 0.001496 seconds.
Elapsed time is 0.000116 seconds.
result 1 = 0.31831
result 2 = 0.31831
```

The sum()*Ts is ~13 times faster than the trapz(t, signal). For a fixed time step, the result is identical.

gbacelli, 2022-08-15:
just a quick note: sum(signal)*Ts performs a rectangular integration, whereas trapz(t, signal) is a trapezoidal integration. As long as the time step Ts is small enough compared to how quickly the signals changes, the two results are close. 
For example, if:
signal = sin(2*pi*200*t);

Then the error is noticeable

gbacelli, 2022-08-15:
here's a better example:
```
n = 50;
x = linspace(0,pi/2,n);
dx = x(2) - x(1);
f_x = sin(x);
trapint1 = (sum(f_x) - (f_x(1) + f_x(end))/2)*dx
trapint2 = trapz(x,f_x)
rectint1 = sum(f_x)*dx
```

results:
```
trapint1 =

    0.9999

trapint2 =

    0.9999

rectint1 =

    1.0159
```

gbacelli, 2022-08-15:
However, when the code gets compiled by creating a function as:
```
function test_int()

n = 50;
x = linspace(0,pi/2,n);
dx = x(2) - x(1);
f_x = sin(x);
tic
trapint1 = (sum(f_x) - (f_x(1) + f_x(end))/2)*dx;
t1=toc;

tic
trapint2 = trapz(x,f_x);
t2=toc;

tic
rectint1 = sum(f_x)*dx;
t3=toc;

fprintf('t1 = %6.6f\n',t1)
fprintf('t2 = %6.6f\n',t2)
fprintf('t3 = %6.6f\n',t3)

disp(['trapint1 = ' num2str(trapint1)])
disp(['trapint2 = ' num2str(trapint2)])
disp(['rectint1 = ' num2str(rectint1)])

end
```
then the performance improvement is a bit different.
```
t1 = 0.000003
t2 = 0.000042
t3 = 0.000001
trapint1 = 0.99991
trapint2 = 0.99991
rectint1 = 1.0159
```

So it may be convenient to manually implement the trapezoidal integration to speed up the calculations and retain accuracy.

gbacelli, 2022-08-15:
a bit faster implementation:
```
function test_int()

n = 50;
x = linspace(0,pi/2,n);
dx = x(2) - x(1);
f_x = sin(x);

w = dx*0.5*[1; 2*ones(n-2,1); 1];

tic
trapint1 = (sum(f_x) - (f_x(1) + f_x(end))/2)*dx;
t1=toc;

tic
trapint2 = trapz(x,f_x);
t2=toc;

tic
trapint3 = f_x*w;
t4=toc;

tic
rectint1 = sum(f_x)*dx;
t3=toc;

fprintf('t1 = %6.6f\n',t1)
fprintf('t2 = %6.6f\n',t2)
fprintf('t4 = %6.6f\n',t4)
fprintf('t3 = %6.6f\n',t3)

disp(['trapint1 = ' num2str(trapint1)])
disp(['trapint2 = ' num2str(trapint2)])
disp(['rectint3 = ' num2str(trapint3)])
disp(['rectint1 = ' num2str(rectint1)])

end
```

```
t1 = 0.000044
t2 = 0.000079
t4 = 0.000016
t3 = 0.000013
trapint1 = 0.99991
trapint2 = 0.99991
rectint3 = 0.99991
rectint1 = 1.0159
```

kmruehl, 2022-08-25:
@johannes-spinneken @gbacelli and @ryancoe, can you explain (in this issue) why the state space formulation is not useful for experimentally derived coefficients? That will help our team better understand the issue and use case. Thank you.

gbacelli, 2022-08-25:
The state space formulation is okay, if you can find a good fit. The problem is due to causality: in particular, the radiation impedance expresses a relationship from velocity to force; that is, the velocity is the input and the force is the output. Physically, however, the causality goes the other way: force (which is a field quantity) is an input and velocity (being a quantity associated with energy) is a state (or output). The result is that it may be difficult to approximate a non causal relationship (radiation impedance) with a causal model (state space model).

ryancoe, 2022-08-25:
A related but distinct reason: the workflow for a linear empirical model is something like:

`time series from sensors` -> `frequency domain spectra` -> `non-parametric freq. domain model` -> `...`

where the `...` includes parametric frequency domain models, impulse response functions, state-space models. The fewer steps you take and, more importantly, avoiding fitting parametric models when not strictly necessary, the better.

[image]

kmruehl, 2023-01-11:
@salhus will lead and @akeeste will support

akeeste, 2023-02-08:
@salhus I've been meaning to work on and get involved in this issue. Can I review the progress here on one of your branches? which branch are you working on?

salhus, 2023-02-08:
@akeeste 
Here is the branch https://github.com/salhus/WEC-Sim/tree/Euler_Fr_conv
I have implemented Euler integration, and alternatively also a discrete FIR block-based implementation.
I need to ascertain the situations where either of the 3 approaches will be prudent.

akeeste, 2023-02-08:
@salhus Thanks for the update. I can run some performance tests on the different cases and see how much improvement is being made.

salhus, 2023-02-08:
@akeeste 
Cheers Adam. I would point out that the choice of simu.dt or the simulink sampling rate does affect the performance, and after increasing the modeling resolution to a certain point, the simulation times virtually converges... (which makes sense)

akeeste, 2023-04-13:
Apologies all involved for the delays. I just got around to reviewing Sal's updates here.

A general note first, based on things brought up in this thread and how they relate to our current implementation--our *current* CIC implementation does not inherit the simulation time, but is only called at intervals of ``simu.CICdt``. This has a couple implications:
- ``interp_factor`` within ``ConvolutionIntegral_interp`` does not need to be variable and can probably be removed entirely. Users can decrease the frequency of the CIC call and increase computational speed by increasing ``simu.CICdt`` in the ``wecSimInputFile.m`` (must be an integer multiple of ``simu.dt``) @johannes-spinneken 
- The implementation isn't affected by variable time steps because it technically doesn't handle the CIC calculation at variable time steps. Regardless if the timestep is variable, fixed or whatever ``simu.dt`` is, the CIC calculation is always called at ``simu.CICdt``

I evaluated the several methods in Sal's branch and brought up in this thread:
- trapz() integration (the current method)
- trapezoidal integration using ``sum()``
- trapezoidal integration using matrix multiplication, as suggested by @gbacelli
- FIR block, as suggested by @salhus 

I ran several tests using MATLAB's ``cputime`` and ``timeit`` commands. The computational time taken, as a percentage of the current implementation is shown in this table. These trends and ratios generally hold for this case (RM3, 2 bodies, body interactions on, dt=0.01s).
[image]

So the FIR filter is the fastest method and improves speed by ~10-15%, but decreasing ``CICdt`` is more significant and readily available. The FIR is probably the most difficult to implement as well, though it should perform better without body-to-body interactions since only a fraction of the blocks are required. Programmatically writing the number of blocks based on number of bodies and b2b is doable, but difficult to set-up and maintain.

@salhus I'm not very familiar with FIR filters. Can you document here why we can use this and any differences between it and the current implementation? I see that the filter ends with a multiplication by ``simu.dt``. I'm assuming this means that variable time steps are not considered. Can you confirm?

The easiest options to improve the CIC computational speed are:
- change the ``trapz()`` call to a matrix multiplication as suggested by @gbacelli [here](https://github.com/WEC-Sim/WEC-Sim/issues/585#issuecomment-1215912265): ``(IRKB*v)*[0.5 ones(n-2,1) 0.5]*dt``
- improve documentation on using ``simu.CICdt``
- consider adding an advanced feature or developer manual section on improving performance time in WEC-Sim

akeeste, 2023-04-19:
Noting items from our weekly discussion that we could also incorporate to solve this issue:
- test the limits of increasing ``simu.CICdt``. Reference theoretical limits of increasing integration steps
- test s-function to increase speed
- @salhus add comments here about FIR implementation
- @salhus will move the FIR implementation into a PR
- @akeeste will add documentation on using ``simu.CICdt`` and general performance improvement to Sal's PR.

akeeste, 2023-05-17:
@salhus Any update on creating a PR from this issue?

salhus, 2023-05-17:
@akeeste 
Sorry for the timelag. I will get to it soon.

akeeste, 2023-06-14:
#1071 helps with this issue

salhus, 2023-09-28:
@johannes-spinneken @ryancoe 
Hi Johannes and Ryan,
The FIR implementation is now merged to WEC-Sim code.  Here is a link to the PR: #1071 
I am closing this issue now. Feel free to re-open this issue if you would like further discussion.

Cheers,
Sal

closed 2023-09-28 by salhus