WEC-Sim/WEC-Sim #587: [BUG]: simulink model fails on initialization

dforbush2, 2021-05-18:
**Describe the bug**
@mmartini1987 wrote: 
I updated WEC-Sim through the "git pull" command, but now I have this error popping up for all my models:

Error using simulationClass/loadSimMechModel (line 106)
block_diagram does not have a parameter named 'SimMechanicsOpenEditorOnUpdate'

Error in wecSim (line 375)
simu.loadSimMechModel(simu.simMechanicsFile); - Show complete stack trace

Can you please give me some support?
 
**WEC-Sim file(s)**
Appears to relate to *.slx file initialization, called from simulationClass.m

**To Reproduce**
[Please provide details on how to reproduce the error]

**Expected behavior**
WEC-Sim runs successfully

**Screenshots**
N/A

**Workstation Information (this section MUST BE completed):**

 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2019b, 2020a, 2020b, 2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v3.1, v4.0, v4.1]

**Additional context**
_Add any other context about the problem here._

dforbush2, 2021-05-18:
@mmartini1987 please populate the template above with more information in the "To Reproduce" and "Workstation Information" sections. I suspect this to be a version issue, possibly related to a switch to 2021a. However, I am unable to replicate your error. Please attach, at a minimum, your wecSimInputFile.m and your *.slx model.

dforbush2, 2021-05-19:
You may try:

>>bdclose all 

Sometimes simulink has problems with consecutive runs if the explorer remains open.

mmartini1987, 2021-05-26:
Thanks @dforbush2 , the 'bdclose all' command worked fine

closed 2021-05-26 by dforbush2