WEC-Sim/WEC-Sim #593: [WEC-Sim Applications] Getting an error while using PTO-Sim

hossein-mehdipour, 2021-05-27:
Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.

Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question
_PTO-Sim_

Describe the bug or issue
_I was trying to use the PTO-Sim, and I followed the exact instructions on _https://wec-sim.github.io/WEC-Sim/man/advanced_features.html#pto-sim_ but I got 2 errors: 
The first one said that there was no oswec.h5 file in the said address, but as highlighted in the screenshot presented below, there is an oswec.h5 file there. And I don't seem to understand what the 2nd error is referring to.
This exact directory works for me when I'm trying to run just wecSim but when I try to run wecSim with PTO-Sim it does not work._
To Reproduce
_Steps to reproduce the behavior:_
1. I used the OSWEC folder `$WEC-Sim\examples\OSWEC`
1. I did the steps described in the _https://wec-sim.github.io/WEC-Sim/man/advanced_features.html#pto-sim_ "OSWEC with Hydraulic PTO"
2. I copied the said files from "OSWEC with Hydraulic PTO" folder `$WEC-Sim\applications\PTO-Sim\OSWEC_Hydraulic_PTO` to my project directory,
3. I also copied the geometry folder `$WEC-Sim\applications\PTO-Sim\geometry` into my directory and replaced the OSWEC example geometry folder
4. I ran wecSimInputFile.m and ptoSimInputFile.m.
5. I typed "wecSim" in the command window.

Expected behavior
_N/A_

Screenshots
[image]

Workstation Information (this section MUST BE completed):
 - OS: [Windows 7]
 - [MATLAB/Simulink Release] [2021a]
 - [WEC-Sim Release] [v4.2]

Additional context
_N/A_

hossein-mehdipour, 2021-06-01:
@kmruehl @jleonqu 
Excuse me for bothering you,
But I thought my problem would have an easy fix (since it's in the very early process of using PTO-Sim),
Isn't there anyone else that can help me, because I was hoping I get guidance sooner so I can move forward with my academic project, but now I'm stuck at this part for the past few days (and as expected I can't find anything online about how to use PTO-Sim except from the WEC-Sim Github page and this Issues page).
Sorry for causing any inconvenience again
Wish you the best of luck

kmruehl, 2021-06-01:
@hossein-mehdipour Thank you for your patience while we respond to this issue. This past weekend was a holiday weekend, for Memorial Day, so everyone has been out of the office for the long weekend. 

Looking at your screenshot and description of your workflow, it appears that the problem is a path issue. While I can see than you have the `OSWEC.h5` file in your `hydroData` directory, that directory is grayed out, which mean it's not accessible in your MATLAB path. That explains why MATLAB is producing an error stating that the file `OSWEC.h5` is missing. There are a few ways to resolve this issue. The quickest would be to add both the `hydroData` and `geometry` directories to your MATLAB path. However, the more robust way to ensure that the `OSWEC_Hydraulic_PTO` example runs as expected is to clone the https://github.com/WEC-Sim/WEC-Sim_Applications repository, and run the https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/PTO-Sim/OSWEC/OSWEC_Hydraulic_PTO application case directly from your local clone.

hossein-mehdipour, 2021-06-01:
@kmruehl Forgive my ignorance, I just found out what Memorial Day is.
And thank you for your guidance, I tried them but unfortunately they were not successful.
I tried the quick way and the `hydroData `and `geometry` directories were no longer grayed out, I still got the same error.
And then I tried the robust way. I've already cloned all the WEC-Sim & its Applications directories from Github, but when I try to run the OSWEC_Hydraulic_PTO it still gives the same error (as shown in the screenshot below). 
I don't if it's important or not but I'm going mention it:
 In order to run the said example I need 2 sets of files, some are from the `WEC-Sim\examples\OSWEC` directory and some are from the `$WEC-Sim\applications\PTO-Sim\OSWEC_Hydraulic_PTO` directory, which causes the copied files (in the example of the screenshot, the `geometry` and `hydroData` directories + afew more files) to show "Not Under Source Control" (their respective circle (under `Git`) is empty not green).
[image]

I would really appreciate it if you could guide me more if possible.

jleonqu, 2021-06-01:
Hello @hossein-mehdipour 

Please follow these steps, they work for me:

1. Make sure you have the latest version of the [WEC-Sim Applications](https://github.com/WEC-Sim/WEC-Sim_Applications) repository.
2. Within Matlab, change your local working directory to: WEC-Sim_Applications\PTO-Sim\OSWEC\hydroData
3. In the command window type 'bemio' to generate the oswec.h5 file needed for the simulation.
4. Now, change the local working directory to WEC-Sim_Applications\PTO-Sim\OSWEC\OSWEC_Hydraulic_PTO
5. In the command type 'wecSim' to run the simulation.

This works for me.

jleonqu, 2021-06-01:
Hello @hossein-mehdipour 

You still have the path issue. In the example, we use “../hydroData”, this will look for the folder up two levels, but your hydroData directory is on the same level, so you should use “./hydroData” instead. The PTO-Sim examples use the same hydroData for two PTO-Sim cases, an thus their relative path is a level up.

hossein-mehdipour, 2021-06-01:
Hello @jleonqu 
Thank you so much for your guidance, it seems that the "." was the problem. I can use PTO-Sim just fine now.
I will try a few more times, and If there was no problems, I will close this issue.
Thank you so much again.
And thank you to @kmruehl for the helpful comments.

kmruehl, 2021-06-01:
@hossein-mehdipour I'm glad that @jleonqu was able to resolve your issue. I'm going to close this issue now.

closed 2021-06-01 by kmruehl

hossein-mehdipour, 2021-07-10:
Hello @joregeleonnnnnnnnnnn
I have a problem about running wecSim using ptoSim in a function and I would really appreciate it if you could help me with it.

So what I'm trying to do is pretty simple, I want to define a function, that gets some input values (of the HPTO) and give the wecSim run outputs as outputs (values like piston pressure & ...)
First I tried to write a script to test my code, it worked perfectly, but when I tried to do the same thing in a function, it would give an error.
It seems that the function can not recognize the ptoSim parameters (at least afew of them),
The screenshot down below shows the function's code and the error.
[image]

And I altered this lines in `ptoSimInputFile.m`:
```
ptosim.pistonNCF.topA = Ap;
ptosim.pistonNCF.botA = Ap;
```
To give you some more information, when I comment out the 3rd line, the function works perfectly. It seems to me that it can't recognize the ptoSim parameters.
I tried something else, I commented out the 3rd line and tried to assign `max(output.ptosim.pistonNCF.topPressure/1e6)` to a variable in `userDefinedFunctions.m` and the same error happened. I wanted to remind you that this exact method works for me when using a Script, but not in a Function. 
I can't seem to find the solution to my problem, I would really appreciate it if you could help me.

hossein-mehdipour, 2021-07-14:
Hello @jleonqu
I wanted to give you an update about my last comment,
I ran wecSim with ptoSim using a function and saved the output structure,
then I realized that it does have a ptoSim struct, but it only has the time matrix and nothing else (nothing else was defined).
I think the problem concerns the workspaces of Matlab, the base workspace and the function workspace and the model workspace.
I even tried comparing the model workspaces when using a function running wecSim vs. when using a script, although they were not identical, I still couldn't find anything useful.
Frankly, I am getting close to my deadline and I would really appreciate it if you could help me solve this problem.

jleonqu, 2021-07-14:
Hello @hossein-mehdipour 

I recommend to use Multiple Condition Runs. You can find the documentation here in this link: [Multiple Condition Runs](http://wec-sim.github.io/WEC-Sim/master/man/advanced_features.html#multiple-condition-runs-mcr)

There are some examples in the WEC-Sim Application Repository here: [Multiple Condition Runs Examples](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Multiple_Condition_Runs)

If you have additional questions regarding this specific topic, I would recommend to start a new issue.