WEC-Sim/WEC-Sim #594: [BUG] Docs not building from #557

kmruehl, 2021-05-27:
**Describe the bug**
https://github.com/WEC-Sim/WEC-Sim/pull/557 docs are not builing

**WEC-Sim file(s)**
N/A

**To Reproduce**
N/A

**Expected behavior**
N/A

**Screenshots**
N/A

**Workstation Information (this section MUST BE completed):**
N/A

**Additional context**
@H0R5E is working on resolving issues with #557 build

H0R5E, 2021-05-28:
OK, it's now building after 4737c71ae751bc05691056b53baf089216c6a182. The travis configuration was still set to try and build the new docs folder and the public encryption keys got overwritten using my versions. I just reverted the relevant files to the state prior to PR being merged.

closed 2021-05-28 by H0R5E

kmruehl, 2021-06-01:
@H0R5E  thank you!