WEC-Sim/WEC-Sim #596: WecSim error with .h5 bemio file

BinoBK, 2021-05-30:
Hello,
Please help me resolve following error that I get when I run wecSim:

********************************
WEC-Sim Read Input File ...   
Error using  < 
Too many input arguments.

Error in wecSim (line 121)
        if h5Info.bytes < 1000

Error in runFreeDecay (line 9)
    wecSim
*************************************
I generated .h5 file with Bemio and was able to generate plots of hydro data. 
I referred to #289, but I am not able to find a solution. I am using the latest wecSim version in Matlab R2020b. 
Kindly advise. 

Regards,
Bino

BinoBK, 2021-06-01:
I am attaching the compressed folder. Path for "bemio.m" file is:  
Trial 6 with Matlab nemoh routine\Folder\hydroData

Regards,
Bino
[Trial 6 with Matlab nemoh routine.zip](https://github.com/WEC-Sim/WEC-Sim/files/6575803/Trial.6.with.Matlab.nemoh.routine.zip)

akeeste, 2021-06-01:
Hello @BinoBK 

In an effort to better address user’s needs, the WEC-Sim team has developed templates for submitting issues. To find the templates open a new issue and choose one of four options: Bug report, Feature request, Theory or implementation clarification, and WEC-Sim applications specific issue or question.

We kindly ask you to reopen this issue using the bug report template. As a development team it is extremely useful for us to classify the issues in these categories and results in solving issues more quickly.

Thanks
Adam

closed 2021-06-01 by BinoBK