WEC-Sim/WEC-Sim #597: [WEC-Sim Applications] Error with .h5 bemio file

BinoBK, 2021-06-01:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
Free_Decay

**Describe the bug or issue**
Nemoh matlab routine was used to generate mesh (Sphere) and hydro data. Free_Decay application template was used for carrying out free decay. bemio.m was used to generate ".h5" file and plots were generated for hydrodata. However during execution of "runFreeDecay.m" code, error is getting displayed as shown in screenshot below. I am attaching the compressed ZIP folder. I referred to #289, but I am not able to find a solution.
[Trial 6 with Matlab nemoh routine.zip](https://github.com/WEC-Sim/WEC-Sim/files/6579811/Trial.6.with.Matlab.nemoh.routine.zip)

**To Reproduce**
_Steps to reproduce the behavior:_

1. Execute 'runFreeDecay.m' in "Folder"

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 10]
 - [MATLAB/Simulink Release][2020b]
 - [WEC-Sim Release] [v4.2]

**Additional context**
_Add any other context about the problem here._

akeeste, 2021-06-01:
Hi @BinoBK 

Thank you for reopening your issue with the template. I looked at your input files and found the issue.

Currently, you are not actually pointing to the .h5 file (or the .stl file), only the directory:
body(1) = bodyClass('..\Folder\hydroData'); 
body(1).geometryFile = '..\geometry';   

You should point to the .h5 file and the stl file and this will solve your issue:
body(1) = bodyClass('..\Folder\hydroData\Folder.h5');
body(1).geometryFile = '..\geometry\spherewomesh.stl';   

Update your input file for each decay case and they should run without issue.

Let me know if you still have problems.

Adam

BinoBK, 2021-06-02:
Hello @akeeste,
Thank you. This solved the problem. 
Regards,

closed 2021-06-02 by BinoBK