WEC-Sim/WEC-Sim #609: Simulation of the WECCCOMP model from GitHub repository does not work for higher sampling time

Liam-Guerrero, 2021-06-09:
**Before all, I know this question has been asked before (The problem of running large Simulation Time-Step about WECCCOMP GitHub repository #582), but the issue is now closed and I do not know how to reopen it or reply from a closed issue.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
WECCCOMP model from GitHub repository.

**Describe the bug or issue**
I need to run the WECCCOMP model for higher sampling times, 25, 50, 75, and 100 ms. however, wecsim arise issues.

**To Reproduce**
1. I read the answer from Nathan on that issue (#582 ).
2. Changed the value of xPcTs = 0.050;
3. Changed the value of simu.dt  = 0.050;    
4. but still getting the same error:
5. I noticed that having the default value for xPcTs = 0.001, and changing just the simu.dt, wecSim "allows" me to increase simu.dt till 0.035. It works well. however if I increase it to 0.036, wecsim crashes. 

**Expected behavior**
Simulation should run for other sampling times.

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10.
 - MATLAB/Simulink Release: 2020b
 - WEC-Sim Version: v4.2

**Additional context**
N/A

nathanmtom, 2021-06-11:
@Liam-Guerrero Thank you for the issue you have raised and I will do my best to try and answer your questions. 

First off, please not that the filters labeled "Differentiator with break frequency" and "First order LP filter" was provided by our collaborators at the University of Aalborg who had previously developed these filters based on their own model development. These were included in the base WECCCOMP model to allow competitors to design their controllers to be incorporated in the experimental test article with the least amount of on site tuning and adjustment. 

Since these filters were designed by the University of Aalborg the WEC-Sim team will not be able to provide much help on how to edit the filter parameters to work with larger time steps. 

Unless you need these filters in your control loop, you can delete them and then use the outputs from the linear PTO block or the revolute joint to get a measurement of the velocity through a different means. 

I hope this response has answered most of your question, but if not please feel free to respond with more information or add clarifying remarks if I have misread the description of your issue.

Cheers, 
Nathan
National Renewable Energy Laboratory

nathanmtom, 2021-06-21:
@Liam-Guerrero I wanted to circle back to see if my above comments have helped solve your issue? If not, let us know what efforts you have made and we can hopefully provide additional guidance and suggestions.

Cheers, 
Nathan
National Renewable Energy Laboratory

closed 2021-06-23 by yuyihsiang