WEC-Sim/WEC-Sim #613: Theory on Frame of reference

timmaflanagan, 2021-06-16:
**Is your question request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I do not understand why I get 'X' result when running 'Y' simulation [...]_
NA

**Describe the theory or implementation approach that you have a question**
_Please clearly describe the assumptions, equations, or code script that you would like clarification. Also, please provide references to other research papers, open-source material, or other information that may be relevant in the WEC-Sim developers understanding your question._
I am trying to determine what frame of reference the forces that WECSim outputs are in.
Specifically: what frame of refence is the PTO force in?
What frame of refence is the Pto velocity?
Are the viscous forces calculated in the body frame or the water frame or the global frame?

**Personal project context**
_If this request is related to your project, please describe what you are attempting to do in your project so the WEC-Sim development team knows your end goal._
NA

**Describe the type of conclusion or resolution you are looking for**
_[Ex.] I am still unclear on how an irregular wave train is generated within WEC-Sim and would appreciate being pointed to where in the documentation and code where I can find additional information._
I am looking to understand what frame of reference the PTO force, and the forceMorrisonViscous is calculated in, so that I can calculate the transmission coefficient of the OSWEC is. If I know the frame of reference, then I can calculate forceMorrisonViscous * velocity (of the body presumably?) = P_losses

**Additional context**
_Add any other context or screenshots about the feature request here._
NA

akeeste, 2021-06-17:
Hello @timmaflanagan 

Thank you for reopening with the issue template.

The forces and response in WEC-Sim are output in body-fixed or pto-fixed reference frames. With regards to your specific questions:
- the PTO force is output in the PTO-fixed reference frame
- the PTO velocity is output in the PTO-fixed reference frame, and is equal to the relative motion between Base and Follower blocks
- the viscous forces are calculated in a body-fixed reference frame

This similarly extended for constraints and moorings output in their own fixed reference frames.

Please let me know if you have further questions on this. I will look at the documentation and work on making the details of the response class more clear.

Adam

timmaflanagan, 2021-06-18:
Hi There,

This is helpful to head, could you please provide the exact mathematical expression used for viscous force (e.g.,  a function of relative velocity between the velocity of the body and the velocity of the water (at CoG depth) expressed in body fixed frame)?

Thanks

akeeste, 2021-06-18:
@timmaflanagan 

These formulations are described in our [theory documentation](http://wec-sim.github.io/WEC-Sim/master/man/theory.html#viscous-damping-and-morison-elements). Please refer to the linked section for detailed expressions. Note that neither the quadratic drag or the linear damping use the relative velocity between the body and the wave. Both use the absolute velocity of the body.

The relative velocity between the body and water is only used with a Morison Element (simu.morisonElement=1,2). In this case, the orbital velocity of the wave is used in addition to the body velocity. 

Also, in the responseClass object these forces are written to output.bodies(#) as:
forceMorisonAndViscous = morison element force + quadratic drag force
forceLinearDamping = linear damping force

timmaflanagan, 2021-06-21:
Thank you for your answer, For the quadratic viscous drag force, is the absolute velocity measured at the center of gravity of the body, or at some random point?

All the best,

Timma

akeeste, 2021-06-22:
@timmaflanagan 

All hydrodynamic forces (including quadratic viscous drag) and the position, velocity and acceleration of the body are evaluated at the center of gravity. 

The one exception to this is the Morison Element force, for which users can prescribe the position of the force application relative to the center of gravity (``body(#).morisonElement.rgME``). If this parameter is set to [0 0 0], then the Morison element force will also be applied at the center of gravity.

timmaflanagan, 2021-06-22:
Perfect!! thank you very much. 
Best,
Timma

closed 2021-06-22 by timmaflanagan