WEC-Sim/WEC-Sim #619: WEC-Sim Testing Master Plan (Proposal)

H0R5E, 2021-06-21:
Following an offline (ish) disussion with @akeeste, I would like to make some proposals for improvements to the WEC-Sim testing processes. The ultimate goal of these changes would be as follows:

> * To have an efficient, high coverage test suite
> * That can be run automatically
> * That could potentially be used for automatic PR merging (for a whitelist of users)

To this end, Adam and I discussed a number of changes, as follows:

1. **Testing the WEC-Sim_Applications repository separately from WEC-Sim**
    - This is probably the most controversial suggestion, but I think there are a number of good reasons
    - For one, the dependency of the WEC-Sim_Applications repository for WEC-Sim is the wrong way around. WEC-Sim_Applications should require WEC-Sim, not visa versa
    - Also, requiring to check-out WEC-Sim_Applications makes the test suite very fragile
    - Moving the WEC-Sim_Applications tests will remove a burdensome load from the WEC-Sim tests (with some caveats, to be addressed below)
    - In the WEC-Sim_Applications repo, each application could be a separate test that can be triggered on a schedule or by pushes to WEC-Sim.
    - Some of the applications tests do provide useful scope for the WEC-Sim integration tests and selected tests should be copied to (and managed as part of) the WEC-Sim code.
1. **Copying of files for testing should stop**
    - Although this is done with good intentions, I think there are a few reasons why it should stop
    - Notably it is difficult (following the copying) to determine what should and shouldn't be checked-into the repo, risking files being added that should not be.
    - It also uses extra space on the user's drive unnecessarily
    - Copying should either be made to temporary space and then deleted or output files should be directed to temporary files and deleted.
1. **The test directory should not be added to the path**
    - Adding the test directory to the path can be risky behavior as it can cause file shadowing
    - A way of conducting the tests without changing the path is required
1. **Clearing the console should be suppressed when testing**
    - If tests fail it is very difficult to determine the cause of failure because the history has been deleted
    - Maybe adding an option for WEC-Sim to run in a quiet mode would reduce the amount of output
1. **Figures should always be closed or not opened at all**
    - Figure generation can be suppressed (see [here](https://github.com/SNL-WaterPower/WecOptTool-MATLAB/blob/master/tests/examples/RM3/ExamplesTest.m) for an example)
1. **Integration tests (i.e. most of the current tests) should be streamlined**
    - These tests should be collected so that certain aspects of the tests can be shared to reduce runtime
    - This will probably involve changes to "inject" a pre-baked solution at some point in the code
1. **Unit tests should be added when ever possible**
    - Increasing unit test coverage will make it easier to determine where a regression has occurred.
    - This process will involve slowing updating the code into smaller unit functions and associated tests
    - This will also encourage writing shorter functions with single responsibility
    - The drawback in MATLAB is that each function needs its own file
1. **Code coverage should be checked and increased if necessary**
    - If some code is untested then new tests will be needed
1. **The wecSimTest function should allow for different options**
    - Such as running the unit tests or integration tests (both by default)
    - Or for choosing particular output (i.e pdf / webpage etc)

Our initial checklist for getting the ball rolling on this is as follows:

- [ ] Adam is going to finish the current set of changes he is making in PR (to follow)
- [ ] Mat will get the test suite running automatically
- [ ] TBD

@akeeste, did I miss anything?

akeeste, 2021-06-22:
@H0R5E This is a good summary of our discussion.

For context, my current CI test PR #620 addresses a few of these items (below). The other major items (1,4,6-8) can be addressed over time as a large ongoing project.

1.  N/A
2. **Copying files**
    - The PR clears all temporary files used to run WEC-Sim simulations during testing. 
    - Files are still copied, but no longer take up space after the tests complete. 
    - .gitignore is updated to ignore these temporary directories and prevent commits with these files.
3. **Test directory**
     - The test directory is removed from the path after testing. File shadowing is prevent by only adding the top level /test/ directory so that the classes can be instantiated.
4. N/A
5. **Figures**
    - The figures from the regression tests are hidden. Right now in the PR they don't appear at all. However, it is not necessary to view them unless the tests are failing and need to be checked. In this case they can be opened manually.
6. N/A
7. **Unit tests**
    - A few unit tests and integration tests for the BEMIO functions are added. Many more are needed, but this is a start.
8. N/A
9. **wecSimTest.m script**
    - The wecSimTest.m script is now a high level function that only setups and runs the tests. Individual tests are left to their respective classes. Also, test output is not longer saved to the 'ans' variable that is easily overwritten. Tests can be properly skipped using the flags in wecSimTest.m that are passed to each test class. 
    - No option to change how the tests are output.

H0R5E, 2021-06-23:
Thanks @akeeste. Note, I just added point #10 to the top post.

akeeste, 2021-11-19:
@kmruehl @H0R5E I went through all of WEC-Sim's functions and noted those that are physics or mathematically based. These functions have a unique solution based on a given input, so it's possible to create a unit test for them. These need to be prioritized of course, but there a lot of functions for which we can make a unit test. 

As an aside, there were some functions (especially those in ``simulink/model``) that are difficult to read and comprehend. I noted those where the documentation/doc strings or variable names could be improved.

[WEC-Sim Functions - proposed unit tests.xlsx](https://github.com/WEC-Sim/WEC-Sim/files/7571774/WEC-Sim.Functions.-.proposed.unit.tests.xlsx)

H0R5E, 2021-11-23:
My coverage data.

[Wec-Sim Coverage.zip](https://github.com/WEC-Sim/WEC-Sim/files/7589588/Wec-Sim.Coverage.zip)

H0R5E, 2022-03-03:
[WEC-Sim_Applications](https://github.com/WEC-Sim/WEC-Sim_Applications) automatic tests on pushes and PRs are now added via https://github.com/WEC-Sim/WEC-Sim_Applications/pull/9 and https://github.com/WEC-Sim/WEC-Sim_Applications/pull/21. I'm now going to start looking at triggering these tests from pushes in this repo.

@kmruehl, I might need to add some direct commits to get this working, but I'll try to avoid it if possible.

H0R5E, 2022-03-04:
OK, so automatic triggering of tests in [WEC-Sim_Applications](https://github.com/WEC-Sim/WEC-Sim_Applications) from pushes to the `master` and `dev` branches of [WEC-Sim](https://github.com/WEC-Sim/WEC-Sim) is now set up. Here is an example of the event for the [master branch](https://github.com/WEC-Sim/WEC-Sim_Applications/actions/runs/1933132100) and for the [dev branch](https://github.com/WEC-Sim/WEC-Sim_Applications/actions/runs/1933402193).

Note that the 20 concurrent runners available with a free account are for **all repositories** owned by the WEC-Sim organisation, so when the WEC-Sim_Applications tests are running these are all taken and runners will not be available for WEC-Sim until the WEC-Sim_Applications tests start finishing.

kmruehl, 2022-03-10:
@H0R5E is this why the tests are run separately and generate individual build reports?

kmruehl, 2022-03-10:
@H0R5E I don't know where I should put this on this PR, but we are not currently testing `wecSimPCT`, and we should add this to the todo list. It should be a pretty simple amendment to the MCR tests

H0R5E, 2022-03-28:
> @H0R5E is this why the tests are run separately and generate individual build reports?

@kmruehl, that is how it is set up, yes. Note that there is an overhead for this as every runner needs to install MATLAB prior to running the tests (and sometimes that doesn't work giving a false negative). Generally, I wouldn't recommend this approach because it uses all the runners up, but given the tests are quite slow, in general, and not run **that** often, I think it's OK to spam the runners to get it done quicker.

I appreciate that it also makes it a little easier to find out where something broke down but, in the end, you still need to look at the log to find out what actually went wrong.  I think something like [test-reporter](https://github.com/marketplace/actions/test-reporter) would be more like what you're looking for, as it produces a nice failure report (in theory). I was going to have a look at this in Q3.

> @H0R5E I don't know where I should put this on this PR, but we are not currently testing `wecSimPCT`, and we should add this to the todo list. It should be a pretty simple amendment to the MCR tests

I think we need to consider wrapping this one up and making a new TODO list or adding something on the project board, perhaps. If we get some coverage reporting added then it will be obvious what is not being tested.

kmruehl, 2023-01-04:
@H0R5E we're going to close this issue, and revisit testing again in the future by putting it on the project board.

closed 2023-01-04 by kmruehl