WEC-Sim/WEC-Sim #623: non-linear excitation force for RM3 point absorber

abaddonwang, 2021-06-22:
Hi, 

I'm doing research on the average power output of a WEC model in different 2D waves. Previously I was using the example model-the RM3 point absorber for the analysis. However, I found the model is a linear model, I want to analyze with a model of high non-linearity. 

I found the examples provided (RM3 point abosorber) is not suitable for the non-linear analysis due to the lack of the mesh file. Can you provide me the mesh file so I can do the non-linear analysis with it (with non-linear excitation force turned on)?

My research topic is mainly on the wave classification, so I'm really weak on model building that's why I come here for help.

akeeste, 2021-06-23:
Hello @abaddonwang 

Thank you for your question. We have begun requiring users to open GitHub issues using an appropriate template. This is very useful to the development team to classify issues, and address questions more thoroughly and efficiently. We hope that this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen this issue with more detail and using the "Theory or implementation clarification" I am happy to help.

Thanks!
Adam

abaddonwang, 2021-06-23:
Hi Adam,

I re-submitted the question in  "Theory or implementation clarification", please check.  I hope can get your reply soon.

Best regards,

Abaddon Wang

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

This email and any files with it are confidential and intended solely for the use of the recipient to whom it is addressed. If you are not the intended recipient then copying, distribution or other use of the information contained is strictly prohibited and you should not rely on it. If you have received this email in error please let the sender know immediately and delete it from your system(s). Internet emails are not necessarily secure. While we take every care, University of Plymouth accepts no responsibility for viruses and it is your responsibility to scan emails and their attachments. University of Plymouth does not accept responsibility for any changes made after it was sent. Nothing in this email or its attachments constitutes an order for goods or services unless accompanied by an official order form.

closed 2021-06-24 by akeeste