WEC-Sim/WEC-Sim #624: [Theory or Implementation] non_linear excitation force turned on using RM3 point absorber model

abaddonwang, 2021-06-23:
Hi,

I'm doing research on the average power output of a WEC model in different 2D waves. Previously I was using the example model-the RM3 point absorber for the analysis. I tried to estimate the influence of non-linear excitation force, I used the example model RM3 point absorber, I turned on the non-linear excitation force by setting 'simu.nlHydro = 2;' in wecSiminputfile. m. However, this will cause the model float away along x-axis during the analysis. 

After consulting with other students, I found I probably need the mesh file for the RM3 model to carry on the non-linear analysis rather than simply turning on the non-linear excitation force by setting 'simu.nlHydro = 2;' 

 Can you provide me the missing file so I can do the non-linear analysis with it ? My purpose is to estimate the difference with and without the non-linear excitation force by using the same wave condition and to see if there is big difference on the power output from the WEC model. (Do you know if this non-linear excitation force will have significant influence on the power output?)

Since my research topic is about the wave field classification, I'm not experienced with model building, so I ask you if you can provide me this model so I can carry one the non-linear analysis. 

Best regards,

Abaddon

akeeste, 2021-06-24:
Hi @abaddonwang 

As stated in issue #623, please use the template when opening a new issue. We ask that you please retain the given headers in bold, and supply information in the given format. This helps the WEC-Sim team to quickly identify the root causes of issues, categorize and analyze issues to find areas where we can improve WEC-Sim. 

For example, please include the following additional information corresponding to the "Theory and Information" headers on the template:
**Is your question request related to a problem? Please describe.**
Does your drift issue happen both with and without nonlinear hydro? What mesh are you currently using? What wave cases are you testing?

**Describe the theory or implementation approach that you have a question**
What requirements for using nonlinear hydro are you looking for clarity on? Is this only about mesh resolution or additional requirements as well?

**Describe the type of conclusion or resolution you are looking for**
I assume you are looking to get the non-linear hydro option working. This may require more insight than handing off a high resolution mesh.

If this template does not fit your needs, we welcome suggestions on new templates or how to improve our current ones.

Thanks!

abaddonwang, 2021-06-24:
Hi Adam,

Sorry, I thought the text is only for guidance. i will re-submit it.

Best regards,

Abaddon

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

This email and any files with it are confidential and intended solely for the use of the recipient to whom it is addressed. If you are not the intended recipient then copying, distribution or other use of the information contained is strictly prohibited and you should not rely on it. If you have received this email in error please let the sender know immediately and delete it from your system(s). Internet emails are not necessarily secure. While we take every care, University of Plymouth accepts no responsibility for viruses and it is your responsibility to scan emails and their attachments. University of Plymouth does not accept responsibility for any changes made after it was sent. Nothing in this email or its attachments constitutes an order for goods or services unless accompanied by an official order form.

akeeste, 2021-06-24:
Thank you! This does help us go through issues more quickly and effectively so that we can help users as soon as possible. I will look to update the template and make this more clear.

abaddonwang, 2021-06-24:
Good evening,

I re-submitted the [Theory or Implementation]non_linear excitation force on RM3 point absorber model #625

Please check.

Best regards,

Abaddon

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

closed 2021-06-25 by nathanmtom