WEC-Sim/WEC-Sim #625: [Theory or Implementation]non_linear excitation force on RM3 point absorber model

abaddonwang, 2021-06-24:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
_Yes, my question is related to the non-linear response of the RM3 point absorber in regular waves, I used this model because it is an tutorial model from the Github, I think it should be where I started._

**Describe the theory or implementation approach that you have a question**
_I tried to run the tutorial RM3 point absorber model using the default regular waves whose wave class is 'regularCIC' and wave height is 2.5m and wave period is 8s, with other setting default, the run is smooth without problem, however,I wanted to find out the influence of non-linear effect on the model, especially the power output. Then I added a line which is 'simu.nlHydro = 2;' in the wecSimInputFile.m, when I run the model again, the result is not sinusoidal anymore,  and from the video from mechanics explorer, the whole model's motion became unstable with time going on, the responses became larger with time,  from about 100s, the whole model floated away as if it was pushed away by the water. After checking, I think this non-linear excitation force should not be turned on by just adding a line, I need to do more work, however, my project is mainly on wave environmental classification and estimation and I'm not good at model building, so I want to ask you if you can provide the RM3 model which can run the non-linear analysis correctly.  I think I need to have the mesh file for the RM3 point abosrber in order to run the non-linear analysis, is that correct?_

**Personal project context**
_My project is about hourly sea states classification, and run these sea states with a WEC model, I need to estimate the influence of non-linear effects on the power output of the WEC model in the same wave condition by running the model with/without non-linear effect turned on_

**Describe the type of conclusion or resolution you are looking for**
_Personally I think the non-linear effect should have an influence on the power output, I don't know if this effect is significant, so I need to compare the power output with/without non-linear effect._

**Additional context**
_Thank you very much for your support_

nathanmtom, 2021-06-25:
@abaddonwang Thank you for reopening issue #624 while following the issues template. 

You are correct that in order to use the nonlinear hydro option in WEC-Sim you will need a mesh of each body in a `.stl` format. The `.stl` file in the `...\examples\RM3\geometry` folder was developed only for animation purposes. The resolution of the mesh `.stl` provided is most likely to coarse to run with `simu.nlHydro=2` which seems to have been confirmed by the simulations you have already run. 

If you have a CAD modeling software like Solidworks, Rhino, or similar, you should be able to take the `.stl` provided and refine the mesh, increase the number of panels, until you no longer have the simulation performance you have described in your issue submission. In the WEC-Sim Applications [Repository ](https://github.com/WEC-Sim/WEC-Sim_Applications) there is an example application for the nonlinear hydro option which has a mesh with sufficient refinement for most operating conditions. Although not the same as the RM3 model, this should provide a target resolution that works for the RM3 model. 

Unfortunately, we have not built a mesh file for the RM3 model that has been checked and verified to run the nonlinear hydrodynamic option. As discussed above, in the applications folder there is an example case with an ellipsoid body that might be adapted to suit your needs. 

I hope this response has answered your question, but feel free to ask for clarification. 

Cheers, 
Nathan
National Renewable Energy Laboratory

abaddonwang, 2021-06-25:
Hi, Nathan,

Thank you very much for your reply and your suggestions, I'm worried that even though I made the refined RM3 non-linear mesh of each body, I still can not know if my model is correct or not. So I'm wondering if you can provide the mesh. stl files. PersonallyI have no experience using the CAD model building software since my research topic is mainly on the wave classification.

I will check the ellipsoid body  model you mentioned.

Best regards,

Abaddon

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

This email and any files with it are confidential and intended solely for the use of the recipient to whom it is addressed. If you are not the intended recipient then copying, distribution or other use of the information contained is strictly prohibited and you should not rely on it. If you have received this email in error please let the sender know immediately and delete it from your system(s). Internet emails are not necessarily secure. While we take every care, University of Plymouth accepts no responsibility for viruses and it is your responsibility to scan emails and their attachments. University of Plymouth does not accept responsibility for any changes made after it was sent. Nothing in this email or its attachments constitutes an order for goods or services unless accompanied by an official order form.

nathanmtom, 2021-06-28:
@abaddonwang Thank you for your response and glad that our initial suggestions appear to have helped. 

Unfortunately, the only `.stl` files available for the RM3 model will be those found in the `examples\RM3\geometry` folder within the WEC-Sim repository. The WEC-Sim team has not developed a `.stl` file, with sufficient refinement, to be used with the nonlinear hydrodynamic option. Apologies, but you will need to build refine your own `.stl` file until it can suit your needs. The nonlinear hydrodynamic option is a useful feature, but does take additional effort to build and verify the model is working accurately based on the environmental conditions simulated. In theory, if the waves are linear than linear WEC-Sim simulation should provide the same output as the nonlinear WEC-Sim simulation because the nonlinear forcing is minimal in these sea states. There is also a limit of how large (or rather nonlinear) the waves can be before the user would need to transition to a high-order modeling software. 

Please let us know if you have other questions or concerns. 

Cheers, 
Nathan
National Renewable Energy Laboratory

abaddonwang, 2021-06-28:
Hi Nathan,

Thank you very much for your quick reply , I think you are right, if the wave is linear, then the linear/non-linear output should be the same.  In my research I have sea states measured from a certain ocean area in Southern UK (Cornwall), these sea states are in the form of hourly 2D spectra S(f). I think right now I need to check if these sea states are linear waves.

Do you know how to estimate the lineararity of waves? I found a figure that can be used to estimate the lineariry of regular waves, see the figure below, it can be seen if the value H/gT^2 is small and d/gT^2 is large, then this regular wave can be considered linear. However,for a 2D wave spectrum with Hs and Tp, in what condition can this wave be considered linear?

[cid:a04b6eb8-e864-40f1-8208-9b4326069cf7]

Best regards,

Abaddon

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

nathanmtom, 2021-06-28:
@abaddonwang Thank you for your update and glad to hear my reply was clear. 

First off, it does not appear that the figure you meant to attach has come through in this issues thread. If possible, can you try to upload the figure again to make sure we know what you are referencing?

Cheers, 
Nathan
National Renewable Energy Laboratory

abaddonwang, 2021-06-28:
Hi Nathan,

Can you see the figure I uploaded this time? I put it in the attachment this time.

Best regards,

Abaddon

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

abaddonwang, 2021-06-28:
[image]

nathanmtom, 2021-06-29:
@abaddonwang Thank you for following up and including the figure in you last post. 

I would agree with your statement that if a regular wave is of small wave amplitude and in sufficiently deep water the wave should be considered linear. 

For a wave spectrum this will be more difficult as the summation of the different linear wave components can lead to varying instantaneous wave amplitudes and periods which could be linear or nonlinear based on the attached plot. If you use the significant wave height and peak period to describe a regular wave and check the above plot and it is linear then I would expect most waves in your sea state would be considered linear. 

Cheers, 
Nathan
National Renewable Energy Laboratory

nathanmtom, 2021-07-06:
@abaddonwang I'm glad that our previous comments made sense and appear to have helped you continue on your research track. 

I wanted to follow-up to see if your questions have been answered so we can close this issue. 

Cheers, 
Nathan
National Renewable Energy Laboratory

abaddonwang, 2021-07-06:
Hi Nathan,

Thank you very much for your concern, I'm analyzing the sea states recently, I will give you a feedback about the results soon .

Best regards,

Abaddon

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

nathanmtom, 2021-07-14:
@abaddonwang We wanted to follow-up to see if you still had an issue? We'd like to close issues if there is no activity for several weeks.

abaddonwang, 2021-07-14:
Hi Nathan,

I checked each of my hourly sea states, there are about 3000 hourly non-directional wave spectra. Among them there are only 5 % of them are inside the linear wave range, the other 95% are within the 2nd order Stokes wave range. The condition of linear (Airy) wave is too strict so the condition is very hard to meet.

I checked the example model RM3 point absorber downloaded from Github, the example wave condition is a regular wave with 2.5m wave height and 8s wave period. And the value of H/gT^2 is 0.004, which is also in the 2nd stokes range from the figure. So I think WEC-SIM must be able to obtain the correct analysis results even in this 2nd order stokes wave condition, isn't it?

You told me in previous email that there is a limit of how large (or rather nonlinear) the waves can be before the user would need to transition to a high-order modeling software. I think WEC-SIM is still OK for 2nd order stokes waves, is that right?

Best regards,

Abaddon

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

nathanmtom, 2021-07-14:
@abaddonwang Thank you for the quick update. Although there are some wave conditions that enter the 2nd stokes range, WEC-Sim assumes the wave profile is represented by a linear (airy) wave profile. Therefore WEC-Sim does not calculate or change the incident wave profile to represent a 2nd order stokes wave. WEC-Sim has been compared to a range of wave conditions that enter the 2nd order stokes wave range and have performed well against experimental results. 

If desired, you can use the `etaImport` wave function to use your own 2nd order stokes wave elevation profile to represent the higher order wave elevation; however, it is not clear if you would see significant differences compared to assuming an regular wave profile. 

I hope this additional information helps in determining your path forward.

Cheers, 
Nathan
National Renewable Energy Laboratory

abaddonwang, 2021-07-14:
Hi Nathan,

I also have a question about the influence of the non-linear excitation force. A friend of mine told me she found when it was turned on the model will drift along the wave direction even with a refined .stl file that allows her to calculate the non-linear excitation force.  (She is using a self-made floating raft model). She found the biggest influence of the non-linear excitation force is on the surge motion along x-axis of the model.

Is the drift motion of the model a nomal phenomenon? The example in the repository (heaving ellipsoid)  only moves in heave so there is no drift phenomenon so I'm not sure if this drift phenomenon represents wrong results.

Best regards,

Abaddon

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

yuyihsiang, 2021-07-14:
So yes, the nonlinear excitation force can be used to calculate the drift force. Several of our papers on focused wave study have shown that the drift force can be capture by using the nonlinear hydrodynamics option, e.g., [1]. Having said that, it is not the most efficient way to calculate the mean drift force. You can also calculate the drift force using the mean drift force option if you only need the mean value [2].  The drift motion is generally a high-order effect. You can approximate it using the mean drift force approach, which only gives you an averaged effect on the drift. 

Here are the papers that you may be interested:
[1] van Rij, J., Yu, Y.-H., and Tran, T. T., 2021, “Validation of Simulated Wave Energy Converter Responses to Focused Waves,” Proc. Inst. Civ. Eng. - Eng. Comput. Mech., 174(1), pp. 32–45.
[2] Ballard, B., Yu, Y.-H., Rij, J. Van, and Driscoll, F., 2020, “Umbilical Fatigue Analysis for a Wave Energy Converter Preprint Umbilical Fatigue Analysis for a Wave Energy Converter Preprint,” 39th International Conference on Ocean, Offshore and Arctic Engineering, OMAE, ed., OMAE.

--Yi-Hsiang

abaddonwang, 2021-07-15:
Hi Yi,

Thank you, I will check the papers.

Best regards,

Abaddon

[1] van Rij, J., Yu, Y.-H., and Tran, T. T., 2021, “Validation of Simulated Wave Energy Converter Responses to Focused Waves,” Proc. Inst. Civ. Eng. - Eng. Comput. Mech., 174(1), pp. 32C45.

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

nathanmtom, 2021-07-21:
@abaddonwang We wanted to check to see if your questions/concerns were resolved now?

Cheers, 
Nathan 
National Renewable Energy Laboratory

abaddonwang, 2021-07-21:
Hi Nathan,

Yes the question has been solved, thank you very much for your support.

Best regards,

Abaddon

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

closed 2021-07-21 by nathanmtom