WEC-Sim/WEC-Sim #628: [BUG] Global Reference Frame in WEC-Sim 'dev' branch

nathanmtom, 2021-06-25:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
In the WEC-Sim `dev` branch when you open the WEC-Sim library, the `Global Reference Frame` block has the following text `Active Method: Custom Parameters` which prevents WEC-Sim from running. I'm hoping to get confirmation that others are experiencing the same issue.  

**WEC-Sim file(s)**
The issue is likely to be in the `WECSim_Lib.slx` library, but further investigation may find a different file that is the issue.

**To Reproduce**
_Steps to reproduce the behavior:_
1. Switch to the WEC-Sim `dev` branch
2. Open the Simulink Library Browser
3. Click on the WEC-Sim Library
4. Double Click on Frames
5. User should see the text stated in the bug description

**Expected behavior**
In the `master` branch the Global Reference Frame block is empty or rather no descriptive text is present.

[image]

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - Windows 10
 - [MATLAB/Simulink Release] 2020b
 - [WEC-Sim Release] v4.1

**Additional context**
The WEC-Sim `master` branch does not appear to have this issue so this should not impact any users using the `master` branch.

akeeste, 2021-06-28:
@nathanmtom 

I did intend for this to default to 'Load Input File' but I might have accidentally saved it with 'Custom Parameters' when merging PR #548 or #604...

I don't get an error using this setting however. What error message appears?

nathanmtom, 2021-06-28:
@akeeste My apologies, I forgot to follow the template and provide the error message. Despite my previous comments, I was also able to successfully run my own WEC-Sim run. I may have previously had the wrong WEC-Sim inputs or model configuration so apologies for the confusion.

Based on this exchange there does not seem to be an issue with the code but the mask for the global reference frame block. Please feel free to close this issue if you have not more questions.

akeeste, 2021-06-28:
@nathanmtom That makes sense, I just wanted to get a fix right away if my Simulink PRs were causing a bug. I'll go ahead and close this issue.

Adam

closed 2021-06-28 by akeeste