WEC-Sim/WEC-Sim #639: OSWEC Rotational PTO Axis

haniyehtp, 2021-07-08:
Hello!
I have some problems running the OSWEC tutorial.
1.	 I see in the geometry file that the flap’s width is in the Y direction, but when I run the OSWEC the coordinates are changed as shown in picture 1. Can you please explain the coordinates system in WEC- SIM and which axis does the rotational PTO rotates around?
2.	Also when I run the OSWEC tutorial and it plays the animation, I see that the flap does not rotate in the right direction that it’s supposed to be the X-axis in picture 1. Instead, it rotates around Y-axis. I tried to change the orientation in the Rotational PTO in Simulink and it came out that the PTO joint is placed at the top of the flap instead of the bottom of the flap as shown in picture 2. The wecSimInputFile is exactly the same as the tutorial.
3.	I see in Global Reference Frame that the seabed is placed at the top of the flap instead of the bottom of the base as shown in picture 3. I have a feeling that it may be the reason why the flap’s joint is placed at the wrong point. Is there a way to change it? Or any other way to make the OSWEC  rotate around the right axis?
Thanks in advance!
Regrads
Haniyeh 
 
[image]
Picture 1

[image]
Picture 2

[image]
Picture 3

nathanmtom, 2021-07-08:
Hello @haniyehtp 

Thank you for your question. We have begun requiring users to open GitHub issues using an appropriate template. This is very useful to the development team to classify issues, and address questions more thoroughly and efficiently. We hope that this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen this issue with more detail and using the "Theory or implementation clarification" we are happy to help. Per the instructions of the issues template please keep the headers for each section of information requested and fill out each section to the best of your availability. 

Please see #625 for an example of the type of information and description our team is looking for.

Cheers, 
Nathan
National Renewable Energy Laboratory

haniyehtp, 2021-07-08:
Hello!
Thank you for your reply. I reopened this issue in #640. Hope it's in the structure that it should be. If not, please inform me to change it.
Regards,
Haniyeh

closed 2021-07-09 by nathanmtom