WEC-Sim/WEC-Sim #640: [Theory or Implementation] OSWEC Rotational PTO Axis

haniyehtp, 2021-07-08:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
I have some problems in running the OSWEC tutorial.
 I see in geometry file that the flap’s width is in the Y direction, but when I run the OSWEC the coordinates are changed as shown in picture 1. I don’t understand the coordinate system in WEC-SIM and how we should model the geometry file in BEM code according to WEC-SIM coordinate system. Because I read in WEC-SIM documentation that the wave direction should be in X direction. So the flap’s width should be in Y direction. But when I run the OSWEC tutorial and it plays the animation, I see that the flap does not rotate in the right direction that it’s supposed to and its shown in picture 1. Instead it rotates around Y axis while the flap and the base width is in X direction. I tried to change the orientation in the Rotational PTO in Simulink and in the animation it came out that the PTO joint is placed at the top of the flap instead of the bottom of the flap as shown in picture 2. I see in Global Refrence Frame that the seabed is placed at the top of the flap instead of the bottom of the base as shown in picture 3. I have a feeling that it maybe the reason why the flap’s joint is palced in the wrong point. Is there a way to change it? Or any other way to make the OSWEC rotate around  the right axis?
**Describe the theory or implementation approach that you have a question**
I didn’t change any information in the wecSimInputFile and the .slx file in picture 1. Though I tried to change PTO orientation in Simulink file to see the results and it’s shown in picture 2.
**Personal project context**
My goal is to model a Langlee WEC and for that purpose I wanted to figure out how the OSWEC tutorial works, so that I can model my project.

**Describe the type of conclusion or resolution you are looking for**
I appreciate if you give me some guidance about the coordinate system in WEC-SIM and how we should assume the coordinate system in BEM code?  And how I can solve the problem of the OSWEC tutorial model to rotate around the right axis?

**Additional context**
[image]
Picture 1
[image]
Picture 2
[image]
Picture 3

Thanks in advance!
Regrads
Haniyeh

nathanmtom, 2021-07-09:
@haniyehtp Thank you for reopening this issue while following the provided issue templates. 

To clarify the statements in your response, the tutorials folder does not have pre-built `wecSimInputFile.m` and `OSWEC.slx` therefore it is possible that in building those two files following the documentation page something might have been coded incorrectly. This is difficult to verify without attaching those two files for review. However, in the WEC-Sim Examples folder, [see here](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/OSWEC), we do have a fully written set of WEC-Sim files that should run the correct orientation you have described above. Therefore, you may want to compare the files you have built through the tutorial with those in the examples folder. If you are willing to share your `wecSimInputFile.m` and `OSWEC.slx` files we might be able to suggest corrections that need to be made.

The WEC-Sim coordinate system is defined [here](http://wec-sim.github.io/WEC-Sim/master/man/theory.html#coordinate-system) and the coordinate system will be the same for your BEM software. The waves can come from any direction you specify but in WEC-Sim the wave heading, \beta, follows the convention in the figure below.

[image]

I hope these answers help clarify you issue, but please let us know if more details or clarification need to be added.

Cheers, 
Nathan
National Renewable Energy Laboratory

haniyehtp, 2021-07-09:
@nathanmtom Thank you, Now I realized I used the example file before, not the tutorial file. Sorry, I wrote it wrong in the description. Here I uploaded the wecSimInputFile.m and OSWEC.slx in the OSWEC.zip file. I did not change these two files in the OSWEC example folder.

[OSWEC.zip](https://github.com/WEC-Sim/WEC-Sim/files/6791753/OSWEC.zip)

Regards,
Haniyeh

nathanmtom, 2021-07-09:
@haniyehtp Thank you for the clarification. Unfortunately, I am unable to reproduce the error you have described when opening this issue. I was able to open the OSWEC example in the WEC-Sim source folder and there was not issue with rotation about axis other than the y-axis. 

In order to try and troubleshoot the issue, I'd encourage you to review the installation procedures and ensure you have successfully added WEC-Sim to the MATLAB path, see [here](http://wec-sim.github.io/WEC-Sim/master/man/getting_started.html#step-1-add-wec-sim-to-matlab-path).

If you believe you have installed WEC-Sim correctly, then there might be a different issue related to your version of MATLAB and WEC-Sim. If possible, can you let us know what versions of each you are running? 

Cheers, 
Nathan
National Renewable Energy Laboratory

haniyehtp, 2021-07-10:
@nathanmtom Thank you, I checked my installation procedure and I guess it doesn’t have any problem. But I’ll try to explain it here:
I added the WEC-SIM folder in the userpath folder. 

'C:\Users\Haniyeh\Documents\MATLAB\WECSIM'

I made an startup.m  file with the following code:

wecSimPath = 'C:\Users\Haniyeh\Documents\MATLAB\WECSIM\source';
addpath(genpath(wecSimPath));

I added this startup.m file in two directories. One in 
'C:\Users\Haniyeh\Documents\MATLAB\WECSIM\source'
And the other one in the Matlab installation directory:
'G:\programs files\R2020b\bin'

Then I wrote the 'path' in the command window and the following path was shown:
C:\Users\Haniyeh\Documents\MATLAB\WECSIM\source

This is how I installed WEC-SIM. If there is any problem in the installation process please let me know. 
Also, my Matlab version is R2020b, and I downloaded master WEC-SIM and I guess its version is v4.2.
I hope there will be a way to fix this problem and I’ll be so grateful If you help me with it.

Regards,
Haniyeh

haniyehtp, 2021-07-11:
@nathanmtom Could you please upload the folder of the OSWEC example that runs properly on your computer? Maybe it can help me. Thanks.

nathanmtom, 2021-07-12:
@haniyehtp Thank you for your response and additional information. 

My first comment is that the startup.m file is likely not in the right location. The first location you have written is in the source folder of WEC-Sim which MATLAB should not look to when initializing paths. The second location appears to be the program files location of MATALB which may work in your local set-up; however, I would try the following: after opening a general MATLAB session [type ](https://www.mathworks.com/help/matlab/ref/userpath.html)`userpath` and based on the path output by MATLAB this will be the location to place your `startup.m` file. 

When you type `path` in the command window is that the only output? The other subfolders should be listed as well. 

The OSWEC example that I ran is pulled directly from the `Master` branch of WEC-Sim that is published on GitHub. Therefore, if you are already having problems with the model from GitHub sending you my local version is likely not to help. I would suggest .zipping your entire working folder to your OSWEC example and attach to this issue to see if any local changes you implemented can be found or if running your folder leads to the issues you have listed above. 

In the meantime, have you been able to successfully run the RM3 tutorial available within WEC-Sim? If this model also works improperly, I would suspect there may be some issues with your MATLAB toolbox installation. 

Cheers, 
Nathan
National Renewable Energy Laboratory

kmruehl, 2021-07-14:
@haniyehtp i just ran the attached OSWEC_example with MATLAB 2020b and am unable to reproduce this error. Please clone (or download) the latest version from WEC-Sim master here: https://github.com/WEC-Sim/WEC-Sim. Thank you!

[OSWEC_example.zip](https://github.com/WEC-Sim/WEC-Sim/files/6819273/OSWEC_example.zip)

haniyehtp, 2021-07-15:
@nathanmtom @kmruehl Thank you, I tried to change the startup.m file path and it worked. Now the OSWEC example works properly. I deleted the startup.m file from the first directory and I just put it in the second directory that I mentioned. I guess the problem was due to the installation problem. Thanks a lot for the support.
Regards,
Haniyeh

closed 2021-07-15 by nathanmtom