WEC-Sim/WEC-Sim #645: Mooring Matrix Stiffness

ckberinger, 2021-07-13:
Hi all,

Can you expand on why you multiply the mooring stiffness by 4 if you have 4 mooring lines? The mooring lines are likely in opposing directions. And in reality, when 2 lines are in tension, the other 2 will likely not be in tension. What is the justification for multiplying by 4?

Courtney

_Originally posted by @ckberinger in https://github.com/WEC-Sim/WEC-Sim/issues/324#issuecomment-879314995_

akeeste, 2021-07-13:
Hello @ckberinger 

Thank you for your question. We have begun requiring users to open GitHub issues using an appropriate template. This is helpful for the development team to classify issues, and address questions more thoroughly and efficiently. We hope that this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen this issue using the "Theory or implementation clarification" template (keep bold headers but can remove italic descriptive text) and with more detail on which case/mooring set-up you're referencing, I am happy to help.

Thanks!
Adam

ckberinger, 2021-07-14:
Hi Adam,

Can you point me to where I can find this template?

Courtney

akeeste, 2021-07-14:
They should appear whenever a new issue is opened on the WEC-Sim GitHub page:

[image]
[image]

Please let me know if they are not appearing and I can look into this.

ckberinger, 2021-07-14:
I think they did not show up for me because I reopened from a comment on an old issue. 
I will reissue using the template, thank you!

akeeste, 2021-07-14:
Thanks for doing that. I will close this one and see if we can set-up the 'reference in a new issue' option with a template.

closed 2021-07-14 by akeeste