WEC-Sim/WEC-Sim #646: [WEC-Sim Applications] can't run WEC-Sim using PTO-Sim in a function

hossein-mehdipour, 2021-07-14:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
PTO-Sim

**Describe the bug or issue**
So what I'm trying to do is pretty simple, I want to define a function, that gets some input values (of the HPTO) and give the wecSim run outputs as outputs (values like piston pressure & ...), but when I do this Matlab gives an error `Unrecognized field name "pistonNCF".`
It seems when I try to run wecSim with ptoSim using a function, then it doesn't recognize the ptoSim parameters and it treats it like a normal wecSim run.

**To Reproduce**
1. Go to `$WEC-Sim_Applications-master\WEC-Sim_Applications-master\PTO-Sim\RM3\RM3_Hydraulic_PTO directory`.
2. Run` bemio.m` in the` $WEC-Sim_Applications-master\WEC-Sim_Applications-master\PTO-Sim\RM3\hydroData`
3. Alter these lines in `ptoSimInputFile.m`:
```
ptosim.pistonNCF.topA = Ap;
ptosim.pistonNCF.botA = Ap;
```
4. Write the function shown in the Screenshot below.
5. type` testFunction (0.18)` in the command window.

**Expected behavior**
I expected this to run smoothly just like when I used a script to do this exact same thing.

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 7]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v4.2]

**Additional context**
To give you some more information, when I comment out the 3rd line, the function works perfectly. It seems to me that it can't recognize the ptoSim parameters.
I tried something else, I commented out the 3rd line and tried to assign `max(output.ptosim.pistonNCF.topPressure/1e6) `to a variable in `userDefinedFunctions.m` and the same error happened. I wanted to remind you that this exact method works for me when using a Script, but not in a Function.
I ran wecSim with ptoSim using a function and saved the output structure,
then I realized that it does have a ptoSim struct, but it only has the time matrix and nothing else (nothing else was defined).
I think the problem concerns the workspaces of Matlab, the base workspace and the function workspace and the model workspace.
I even tried comparing the model workspaces when using a function running wecSim vs. when using a script, although they were not identical, I still couldn't find anything useful.

hossein-mehdipour, 2021-07-14:
Hello @jleonqu 
I started a new issue just as you said. 
About your last comment, I know what MCR is, but I don't seem understand how it would help my situation. I'm trying to something different. I don't have all the values of different parameters of the run from the start, It is an optimization problem, I would give the function one input, it gives me an output, then I give the function another input accordingly and ... .
I really hope that you can help me solve this problem.

H0R5E, 2021-07-14:
Could be fixed by #632

hossein-mehdipour, 2021-07-15:
@H0R5E Thank you so much, that was exactly my problem and the solution I was looking for.

closed 2021-07-15 by hossein-mehdipour