WEC-Sim/WEC-Sim #647: [Theory or Implementation] Mooring Matrix Number of Mooring Lines

ckberinger, 2021-07-15:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
N/A

**Describe the theory or implementation approach that you have a question**
I am developing a mooring matrix for a point absorber WEC. The physical device will have 4 mooring lines in an x shape.
Thie issue below addressed this setup by suggesting to multiply the mooring stiffness by 4 if you have 4 mooring lines. 
[https://github.com/WEC-Sim/WEC-Sim/issues/324]

And this issue references this physical test of the RM3.
Yu, Y., Lawson, M., Li, Y., Previsic, M., Epler, J., & Lou, J. (2015). Experimental Wave Tank Test for Reference Model 3 Floating- Point Absorber Wave Energy Converter Project. NREL Report, (January).

I looked through the mooring matrix documentation, source code, simulink, applications/examples, and webinars and didn't see anything that expanded upon multiplying the mooring stiffness by 4 for 4 lines. 

**Personal project context**
When considering a physical setup of 4 mooring lines with pre-tensioned springs in an x configuration (as was in the paper above) wouldn't only 2 mooring lines be used to return the device to the original position? And the other 2 would either not be in tension or not pulling it back to the original position? This is referencing displacement in the x direction with unidirectional waves. 

**Describe the type of conclusion or resolution you are looking for**
I am looking to understand how the mooring matrix works and understand the justification for multiplying the mooring stiffness by 4 when the mooring lines are not inducing forces in the same direction.

**Additional context**
N/A

akeeste, 2021-07-20:
Hi @ckberinger 

Thanks for reopening with the issue template, I will do my best to address your question and reference this figure:
[image]

For the first part of your question: 
- **When considering a physical setup of 4 mooring lines with pre-tensioned springs in an x configuration (as was in the paper above) wouldn't only 2 mooring lines be used to return the device to the original position? And the other 2 would either not be in tension or not pulling it back to the original position? This is referencing displacement in the x direction with unidirectional waves.**

This comes down to the assumptions that are being made about the mooring lines. When treating the mooring line as a spring-damper matrix, there is no slack. The springs are always in tension, compression or at their equilibrium length. In your example, given a +x displacement lines 1,2,4 are in tension. Line 3 must be in compression because no slack is allowed. This is why a single net stiffness/damping matrix is used, regardless of the device's direction of motion. 

If the mooring model needs to include weight and slack within the lines, then I recommend using MoorDyn. I have not used MoorDyn much so I can't advise on using it, but it should account for the slack, weight, stiffness, and direction of the lines, etc.

For the second part of your question: 
- **I am looking to understand how the mooring matrix works and understand the justification for multiplying the mooring stiffness by 4 when the mooring lines are not inducing forces in the same direction.**

The opposing springs are inducing forces in the same direction. For example, given a motion of +dx:
Line 1 is in tension and applies a force in -x to the WEC, 
Line 3 is in compression and applies a force in -x to the WEC

You will find the same result if considering what the spring 'sees' in the local frame and convert to the global coordinate system.

In general if modeling the mooring lines as a stiffness/damping matrices is sufficient, then you can calculate the net matrix of all lines. However I would edit the referenced response and suggest that you sum the mooring matrices of each line, rather than multiply them by 4. For example, if stiffness is large in +x and small in +y (in a line's local frame), the stiffness and damping matrices of lines 2,4 will be rotated 90deg from lines 1,3 and results in large stiffness in +y and small in +x (in global frame)

Please let me know if this answers your question or if any part of the explanation is unclear.
Adam

ckberinger, 2021-07-23:
Hi Adam,

Thank you for explaining and including the image. The physical mooring set up in the referenced paper cannot be in compression (based on the cables connected to the springs) so that was my main concern with multiplying the mooring stiffness by the number of lines. This answers all my questions.

Courtney

akeeste, 2021-07-26:
Hi Courtney,

Okay I understand the confusion now. In this case, I'd recommend MoorDyn for the mooring system, rather than the mooring matrix.

closed 2021-07-26 by akeeste