WEC-Sim/WEC-Sim #648: [WEC-Sim Applications]

rohitalexander, 2021-07-16:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
PTO-Sim

**Describe the bug or issue**

I am working on a project for my M.Sc dissertation. I was looking into if I can implement PTOSim for my project. While I was trying out the RM3 tutorial using PTOSim, I noticed that I am having the same issue mentioned in #216 . The PTO Library does not contain the "Translational PTO UD Force (Local Z)" block. It was suggested in #216 to ensure the latest version is being used. I think I am using the latest WEC-Sim version v4.2 (I downloaded the source code from : https://github.com/WEC-Sim/WEC-Sim).

**To Reproduce**
_Steps to reproduce the behavior:_
N/A

**Expected behavior**
N/A

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: macOS Big Sur 11.4
 - [MATLAB/Simulink Release]: MATLAB_R2020b
 - [WEC-Sim Release]: WEC-Sim v4.2

**Additional context**
N/A

akeeste, 2021-07-20:
Hi @rohitalexander 

Thank you for opening with the issue template. It appears you have the correct, up to date version of WEC-Sim. Are you following the documentation directions [here](http://wec-sim.github.io/WEC-Sim/master/man/advanced_features.html#tutorial-rm3-with-pto-sim)? It seems our documentation for setting up the PTO-Sim example is quite outdated. 

We no longer have a ``Translational PTO UD Force (Local Z)`` block. Instead use the block ``Translational PTO Actuation Force``. 

These blocks were previously simplified into a single block, whose direction can be arbitrarily set with ``pto(#).orientation``. The PTO will translate in the z direction, which by default is [0 0 1]. For this example, you can use the ``Translational PTO Actuation Force`` as is without changing the orientation.

Please let me know if this solves your issue. I will leave this issue open until the documentation is updated by a PR.

Adam

rohitalexander, 2021-07-20:
Hi Adam,

Thank you for clarifying. This does solve my issue.

Many Thanks,
Rohit

closed 2021-07-23 by nathanmtom