WEC-Sim/WEC-Sim #650: [Theory or Implementation] Differences in power outputs - Regular and Irregular waves

rohitalexander, 2021-07-20:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
I am working on a project for my M.Sc. dissertation on a simple cylindrical buoy WEC operating in heave. I used NEMOH as the BEM solver. I tried to evaluate the power absorbed by the buoy in regular waves and irregular wave (Jonswap spectrum) both with significant wave height of 3 m and peak period of 10s. While I do obtain results, I can see a big difference in the power outputs for both the conditions, and this had me thinking if I am missing something here. I executed NEMOH for just one wave direction. Do I need to have multiple wave directions to implement the Jonswap spectrum? Could that be a reason I am getting inconsistent result?

**Describe the theory or implementation approach that you have a question**
_Please clearly describe the assumptions, equations, or code script that you would like clarification. Also, please provide references to other research papers, open-source material, or other information that may be relevant in the WEC-Sim developers understanding your question._
N/A

**Personal project context**
_If this request is related to your project, please describe what you are attempting to do in your project so the WEC-Sim development team knows your end goal._
A simple evaluation of the performance of the wave energy converter and obtain forces acting on the WEC as well as the power absorbed/generated by the WEC. 

**Describe the type of conclusion or resolution you are looking for**
_[Ex.] I am still unclear on how an irregular wave train is generated within WEC-Sim and would appreciate being pointed to where in the documentation and code where I can find additional information._
I am not sure if there should be huge difference in the power outputs when subjected to regular or irregular waves. 

**Additional context**
_Add any other context or screenshots about the feature request here._
[image]
[image]
[Test-Heave.zip](https://github.com/WEC-Sim/WEC-Sim/files/6850631/Test-Heave.zip)

Many Thanks,
Rohit

akeeste, 2021-07-20:
Hi @rohitalexander 

Thanks for using the template and uploading your case file. 

The Jonswap spectrum only requires BEM data across various frequencies, not different directions. A 1-direction run at across the relevant frequency range is sufficient.

This seems like a reasonable comparison between regular and irregular waves. The average power for each case appears to be about the same, ~``-1.75e6 `` The irregular wave case has a much larger peak power, but this seems reasonable given that the wave height from 175s-250s is likely larger. If you compare the wave elevations for both cases, I think you will see that the increased wave elevation corresponds to the increased PTO power.

Adam

rohitalexander, 2021-07-20:
Hi Adam!

Thank you for your response. That does makes sense. I did observe that the wave elevation from ~ 175s to 250s were higher, but wasn't sure if it would correspond to a much higher power output. But yes, the average output appears to be same. So, as such can I assume that my approach is correct and I do not need to define anything in the BEM part to implement Jonswap spectrum in WEC-Sim?

Many Thanks,
Rohit

akeeste, 2021-07-26:
Hi Rohit,

Sorry for the delay on your last question. I did not dive into your BEM data much, but yes it sounds like you are implementing the JS spectrum correctly. Let me know if there is anything else on this topic, otherwise I can close out the issue.

Adam

rohitalexander, 2021-07-26:
Thank you, Adam. That's all for now.

Kind Regards,
Rohit

closed 2021-07-26 by rohitalexander