WEC-Sim/WEC-Sim #653: Use of rot. PTO act. torque/rot. PTO act motion for pos/speed control

t-osu, 2021-07-27:
**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
N/A (maybe PTOs)

**Describe the bug or issue**
Constant speed/position control with PTO actuators. Cannot get to move in constant behavior. Period is 5 seconds, expect speed to be 1.256 rad/s.

**To Reproduce**
I don't know if this can be applied to my case (someone reproducing it) since it is using my own hydro bodies and not some from an example, but I think the screenshots below will get the idea across. I guess you could recreate it with the OSWEC example and add the various actuators shown in each separate setup, but mine is a pendulum and I want it to continuously make full revolutions in yaw whereas the OSWEC doesn't make full revolutions in pitch, but here's what you would do for the 3 cases:

1) rotational PTO actuation motion block. I simply put the speed I want at the velocity input. This is the model what is outputted as speed. screenshot number 1 and 2
2) position for the pendulum to follow using a repeating sequence (I didn't want to use a sine wave because then the pendulum wouldn't be moving in one direction). screenshot number 3 and 4.
3) rotational PTO actuation torque. Demux the response output from this block to get the velocity of the pendulum which is then put into a simple PI controller where the reference is the speed 1.256. To get to torque, just multiply it by a coefficient. I'm unsure how to determine that currently for different speeds because it obviously has an effect on the overall speed I am actually getting. Right now, just have it set to a value that got me on average the closest to my desired speed. I didn't use the direct drive rotary generator from PTO-Sim because of all the parameters needed to be set and I'm unsure what my actual generator parameters would be. screenshot 5 and 6.

**Expected behavior**
1) Expected to move at constant speed since a constant is the input. The output confuses me because if it is going at that speed, why is the position not at a constant slope? (I have the position unwrapped.) When watching the model in the model explorer, the pendulum is moving very slowly, definitely not the speed it should be.

2) Expected to move at constant speed since has position to follow at specific times that it should be completed in. The pendulum is moving in one direction, but the speed continually increases, though.

3) Expected a constant speed with the PI controller, this one does the best, but is still jumpy around.

**Screenshots**
1) 
[image]

2)
[image]

3)
[image]

4)
[image]

5)
[image]

6)
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Microsoft Windows 10 Enterprise Version 10.0 (Build 19042)
 - [MATLAB/Simulink Release] MATLAB Version: 9.9.0.1592791 (R2020b) Update 5...Simulink Version 10.2
 - [WEC-Sim Release] v4.2

**Additional context**
Any ideas on why I can't achieve a constant speed? Thank you!
- Chris

akeeste, 2021-07-27:
Hi @t-osu 

Thank you for reopening with the issue template. I will do my best to address your question.

On your various PTO set-ups:
If you do not yet know your motor characteristics, I agree that the PTO-Sim examples are more complicated than necessary right now. I recommend using set-up 1. where you define the velocity input. However, you should also input the appropriate acceleration and position along with the velocity. In this case, input a constant 0 acceleration and input the integrated velocity for position. Do not input position and use the derivative block for velocity and acceleration as this will add noise to your system.

On the velocity plots:
This seems like a reasonable velocity to prescribe in yaw. Based on your comment in screenshot 5, I assume your pto orientation is set correctly (``pto(#).orientation.y = [0 0 1]``).
What do you mean by "you have the position unwrapped"? Also can you clarify which response you are plotting, the body or pto response?

The PTO response is relative between the connected bodies. So, if you are prescribing some motion in yaw the pendulum should move relative to body(1) in that way. However, if body(1) also has motion in other dofs (e.g. in surge, heave or yaw) then body(2) will also contain that velocity component. Is this causing the discrepancy in your plots?

Lastly, can you upload your case file if possible? That will help me debug your issue.

Thanks,
Adam

t-osu, 2021-07-28:
Morning @akeeste,

Thanks for the suggestions, I was unsure if all pos/vel/acc ports needed to be filled or not. I did as you said and applied a constant 0 to the acc and integrated the velocity. The position plot is as it should be, a constant slope line, but the velocity is still jumping around before settling to a value larger than what is being inputted as you can see in the screen shot below. 

[image]

The PTO is oriented to rotate in yaw, correct.

Wrapping:
Instead of the position plot stopping at -pi and then having a discontinuity and jumping back to +pi, I unwrapped it so that you can see it moving in a constant accumulating direction. Makes it easier to see the constant slope.

Plots:
I am plotting the pendulum bodies position and velocity, and the only DOF constraint on it is the rotational PTO in yaw, but yes, body 1 is moving in all 6 DOF. So since those velocities will have an impact, maybe that is the extra velocities I am seeing? I guess I was unsure how strong the PTO actuator was in controlling the behavior of the follower in respect to the base.

Sorry, I don't think I can upload it, screenshots are the best I can do!
Thanks,

- Chris

akeeste, 2021-07-28:
Hi Chris @t-osu 

Thanks for clarifying the above points. I understand about the case. I think we can debug this anyways.

I recommend plotting the output PTO position as well as both bodies' motion and the relative body motion. You should see that the relative body motion is identical to the PTO motion. Then you can confirm if the PTO motion matches your input signal or not.

If this is not robust, it seems there are a few options you can change within the [Simscape joint ](https://www.mathworks.com/help/physmod/sm/ref/prismaticjoint.html) itself. The linked documentation references a high/low priority option that can better enforce the motion inputs. I have not tested the limits of prescribing a motion, but in my experience Simscape should be able to handle the input unless it requires extremely high forcing.

t-osu, 2021-07-28:
No problem.
Good suggestion. I did it and they are not the same and move away from each other later. 

[image]

I actually have tried the state targets in the past and it did not do much. I did just check now the pto velocity and it is a constant 1.256 rad/s the whole time. Here it is plotted with the body velocity. Unfortunately, I can't put in a position target because it needs to be a scalar.

[image]

I will check other simscape blocks, but the options in the joint block seem to be the only way to influence the follower.

akeeste, 2021-07-29:
Hi @t-osu

The body and PTO motion are not necessarily the same. The pto(1) velocity should be the same as the _relative_ velocity between body(2) and body(1). However it appears you are plotting the velocity of a body and the pto.

Also, as I cannot run your case please be very clear on what is being plotted. In your plots, please include the degree of freedom in the title, the body number in the legend and axes labels for my reference.

Can you clarify, are you inputting a position target to pto(1) now? I thought you were after our first exchange but am uncertain. What do you mean about the position target needing to be a scalar?
**I did as you said and applied a constant 0 to the acc and integrated the velocity.** vs 
**Unfortunately, I can't put in a position target because it needs to be a scalar.**

t-osu, 2021-07-29:
Ah, yes, I read one of your comments too fast. Right, relative velocity. 

Let's backtrack. Here is the setup on the outside: a constant velocity, integrated for position, then a constant block with 0 for acceleration.

[image]
Here is a plot of the velocities again with a clearer legend. The relative velocity is in orange-red, but it is behind the purple line because the difference between body 1 and body 2 velocities is small since body ones velocity is small:

[image]

I then went into the joint revolute block inside the PTO block and specified a velocity state target, upon doing this, I did not get a better output than above.
[image]

So I tried to enter in addition to a velocity state target a position target to make it more "robust", but the position is not a scalar, it would have to be an array of values because the pendulum is constantly moving. This doesn't work though, for example, if I put the array [0 1 2 3] as a target, I get the following error which makes sense to me, it having to be a scalar. I just wanted to try using both since the velocity one didn't work:
[image]
[image]

Now I am back to just the velocity state target. 
After all these good chats, I think I may have been misinterpreting the block. So, the PTO actuator is for the PTO motion itself, not for the follower? In other words, it will make the PTO be a constant velocity, and the relative velocity between the bodies will adjust to make this condition true? If that is correct, it is odd that the relative velocity in the plot above does not match the PTOs velocity.

Let me know if this is helpful or if I need to clarify more things! Thanks.

akeeste, 2021-07-29:
Hi @t-osu 

Thank you for clarifying. That is helpful for me to understand.

On the state targets--
Sorry for my confusion on the state targets vs the PTO inputs. I thought the actuation input (response in this case) would effectively be the state target, however I see now that Simscape treats them differently. I agree that only the velocity state target could be used in this case, as the position will be changing with time.

On the PTO-- 
Yes the PTO actuation motion is for the relative motion between the connected bodies, not the joint itself. We do not actually input or output the motion of the joint itself. You can view this in the Mechanics Explorer, but WEC-Sim does not use this data. 

The PTO motion (whether an input or output) always refers to the relative motion between the two bodies. When using PTO actuation motion, the follower's response is determined by the combination of the PTO input (relative velocity) and the motion of the base. Simscape then determines the constraint force required to enforce the prescribed motion.

For the figure, I would double check your plotting scripts. It appears that if you take ``body 2 yaw velocity - body 1 yaw velocity``, the result is identical to the PTO velocity as it should be.

I hope this explanation of the PTO is helpful. I made a note on the project board to better detail what the response class contains. I am interested to hear if you have suggestions on the best documentation section to make these additions. Maybe API, code structure, or advanced features?

Adam

t-osu, 2021-07-30:
All good.
You're right the velocities match, I just accidentally had `output.bodies(2).velocity(:,6) - output.bodies(1).velocity(6)` and forgot the colon to include all rows so it was only subtracting from one value! I fixed that and the relative velocity lines up with the PTO velocity :]

OK, so I guess there is no way for me to make a _body_ move at a certain speed I want? 

That was just an honest overlook on my part in remembering what the PTO represents, but I would put it in advanced features. My personal go-to is the API, but I think there might be better information intake to a user if it were in advanced features.

akeeste, 2021-07-30:
@t-osu 

Thanks for your input on the documentation! I'm glad I could help clear things up and that WEC-Sim is working properly.

It will be difficult to de-couple the pendulum motion from the base since they are attached. In order to more precisely define the pendulum motion in the global frame, there will need to be better control over the base motion. Maybe use additional mooring stiffness/damping to decrease the base motion in yaw, or even constrain that DOF entirely. Alternatively you could create a controller that takes in the base motion, and prescribes a relative motion between the bodies to give the desired pendulum motion. I can't say whether this would represent the actual system well, but that's where I would start for the simulation.

Let me know if there's any other item related to the WEC-Sim PTO I can clarify.

Adam

closed 2021-07-30 by t-osu