WEC-Sim/WEC-Sim #656: [Developer Issue]

akeeste, 2021-07-29:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Project**
WEC-Sim development

**Describe the issue**
The output time step ``simu.dtOut`` is not robust in variable step WEC-Sim simulations. Library ``To Workspace`` blocks currently use a [decimation ](https://www.mathworks.com/help/simulink/slref/toworkspace.html) of ``round(simu.dtOut/simu.dt,0)``. For fixed step solvers this is fine, but with variable step solvers this becomes inaccurate. For example, if simu.dt = 1, simu.dtOut=5, WEC-Sim outputs every 5th data point. However when using a variable step solver, the actual time step can become 0.01. Then the output time step is effectively 0.05, not 5 seconds.

This can be fixed by changing all ``To Workspace`` blocks in the library to use the ``SampleTime = simu.dtOut``

dforbush2, 2021-08-04:
see also global model settings: 
[image]

not sure which is the easier/better way

akeeste, 2021-08-04:
@dforbush2 Thank for pointing this out. I'll look at loading the simulink file with the global output time step instead of changing each individual ``To Workspace`` block.

commit a645fa8a (2021-09-15): Bug fixes for variable step solver resolves #656 (#702)

* change to workspace blocks decimation to sample time

* clean up bemio.m clears

* update wave elevation calculations for variable step solver

* update simu.time to be actual simulation time
closed 2021-09-15 by kmruehl