WEC-Sim/WEC-Sim #662: Required information on WAMIT's .out file

rbeicht, 2021-08-06:
Hello everyone,

I'm using WEC-Sim for a new project with some collegues, and after reading the documentation and old questions here, there's still a doubt. 

From what I read, when using WAMIT files in BEMIO, the .out needs to consider all 6 DoFs, and also have the infinite added mass. Then, I would like to know if it has any other requiriment, or just those 2.

Thanks for the help!

akeeste, 2021-08-06:
Hello @rbeicht 

Thank you for your question. The WEC-Sim development team has begun requiring users to open GitHub issues using a template, as described in our [Troubleshooting documentation](http://wec-sim.github.io/WEC-Sim/master/user/troubleshooting.html#troubleshooting). This is helpful for the development team to classify issues, and address questions more thoroughly and efficiently. We hope that this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen this issue using the appropriate template (keep bold headers but remove italic descriptive text) and linking to the relevant documentation that you have questions on, we are happy to help.

Thanks!
Adam

rbeicht, 2021-08-06:
Thanks for the advice mate, done it!

kmruehl, 2021-08-06:
@rbeicht yes, those are both requirements for WAMIT output in order to run WEC-Sim. WEC-Sim does require data for all 6 DOF from WAMIT for each body, and the infinite added mass. Here's a list of information BEMIO needs to parse WAMIT data for a WEC-Sim run:

- frequency-dependent excitation force 
- frequency-dependent radiation damping 
- frequency-dependent added mass and infinite added mass
- hydrostatic restoring force 
- displaced volume

We have [examples of WAMIT runs for WEC_Sim here](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/BEMIO/WAMIT) if you'd like to refer to prior WAMIT runs when configuring your new run. Personally, I always start with a prior run and modify the input files for a new case.

rbeicht, 2021-08-09:
> @rbeicht yes, those are both requirements for WAMIT output in order to run WEC-Sim. WEC-Sim does require data for all 6 DOF from WAMIT for each body, and the infinite added mass. Here's a list of [...]

Thanks for the info mate, it will help a lot!

Only thing I didn't understand was modifying input files in new case, you mean to use everything like the example but change only WAMIT file?

akeeste, 2021-08-16:
@rbeicht This comment was only to suggest that if you base the WAMIT configuration files (.cfg, .frc, .pot) off of the BEMIO examples, then all information necessary for WEC-Sim should be included. Then when creating a new WAMIT case, you only have to edit a few parameters (cg, frequency, headings, etc). The rest of the input files stay the same and ensure all outputs for WEC-Sim are included.

Does this help resolve your issue?
Adam

rbeicht, 2021-08-17:
Oh I see, now I understood! 

It certainly helps and also is a good strategy to avoid mistakes or missing outputs. Thanks a lot!

closed 2021-08-17 by rbeicht